espressif/esp-idf #4798: Regarding paho mqtt library for esp-wroom 32 module

manojsai17, 2020-02-18:
please provide the paho mqtt library working for esp32 module .

I included the paho library  , but i am getting length of packet . but it s not forming the mqtt connect packet , mqtt publish packet .

please provide the suggestion or any library link for paho mqtt with esp-wroom 32

david-cermak, 2020-02-25:
Hi @manojsai17 

There is currently no plan to add paho mqtt library into IDF. In case you are porting the paho library to ESP platform and experiencing some specific IDF related problems please post the errors here. I'm afraid we cannot help with such a general issue otherwise.

projectgus, 2020-03-02:
Hi @manojsai17,

For a supported way to use MQTT on ESP32, take a look at the ESP-MQTT library included with ESP-IDF:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mqtt.html

Regarding the Paho library, if you choose to port it then as David says we're happy to help with any specific problems that might be issues in ESP-IDF itself.

closed 2020-03-02 by projectgus