espressif/esp-idf #4800: Spamming 'wifi: Haven't to connect to a suitable AP now!'

vonnieda, 2020-02-18:
## Environment

- Development Kit:  none
- Module or chip used: ESP32-WROVER-B
- IDF version: v3.3.1
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: macOS
- Power Supply: USB

## Problem Description

This morning I noticed that one of my boards was spamming the message 'wifi: Haven't to connect to a suitable AP now!'' on the console. For example:

```
W (40478) wifi: Haven't to connect to a suitable AP now!
W (40598) wifi: Haven't to connect to a suitable AP now!
W (40708) wifi: Haven't to connect to a suitable AP now!
W (40828) wifi: Haven't to connect to a suitable AP now!
W (40948) wifi: Haven't to connect to a suitable AP now!
W (41068) wifi: Haven't to connect to a suitable AP now!
W (41188) wifi: Haven't to connect to a suitable AP now!
W (41298) wifi: Haven't to connect to a suitable AP now!
W (41418) wifi: Haven't to connect to a suitable AP now!
W (41538) wifi: Haven't to connect to a suitable AP now!
W (41648) wifi: Haven't to connect to a suitable AP now!
W (41768) wifi: Haven't to connect to a suitable AP now!
W (41888) wifi: Haven't to connect to a suitable AP now!
W (42008) wifi: Haven't to connect to a suitable AP now!
W (42118) wifi: Haven't to connect to a suitable AP now!
W (42238) wifi: Haven't to connect to a suitable AP now!
```
I have power cycled the board and have reflashed it several times and this persists. It started after I purged all connectable wifi networks from the device - thus esp_wifi_connect() is not being called.

This is a situation I test regularly, and have tested as recently as last night. 

My WiFi initialization is as follows:

```
event_group = xEventGroupCreate();
xx_ERROR_CHECK_RET(esp_event_loop_init(event_handler, NULL));
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
xx_ERROR_CHECK_RET(esp_wifi_init(&cfg));
xx_ERROR_CHECK_RET(esp_wifi_set_storage(WIFI_STORAGE_RAM));
xx_ERROR_CHECK_RET(esp_wifi_set_mode(WIFI_MODE_STA));
xx_ERROR_CHECK_RET(esp_wifi_start());
xx_ERROR_CHECK_RET(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
```
After this initialization I start a task which performs site scans and attempt to connect.

I have disabled the task entirely, and commented parts of the initialization. The warning starts as soon as esp_wifi_start() is called, even if I never perform a scan or attempt esp_wifi_connection().

I have searched the ESP-IDF code for this error and not found it, so I assume it is part of the WiFi blob.

Can someone tell me what state causes this warning to appear?

Thanks,
Jason

gecko242, 2020-02-19:
Seems very similar to #4778 that I reported recently...

zhangyanjiaoesp, 2020-02-21:
@vonnieda This log shows that you called the function `esp_wifi_get_ap_info()` before you had connected to an AP.  If you want to get the scan results and find an AP to connect, you need to call function `esp_wifi_scan_get_ap_records()`

vonnieda, 2020-02-21:
Hi @zhangyanjiaoesp, thank you for the response. This lead me to find all instances where I am calling esp_wifi_sta_get_ap_info and indeed, there is one case where I call it often enough to explain this behavior. Thanks!

closed 2020-02-21 by vonnieda