espressif/esp-idf #4804: compilation fails on 4.0 example

korstiaanS, 2020-02-19:
Hi,

I just tried the latest stable version 4.0 and took the wifi example station.
During compilation however I receive following errors and cannot compile:

```
In file included from /Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired, order)
                               ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
          ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:387:2: error: incompatible type for argument 1 of '__atomic_exchange_n'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
  ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_clear_explicit':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:285:28: error: request for member '__val' in something not a structure or union
  __atomic_store_n(&(object)->__val, desired, order)
                            ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:394:2: note: in expansion of macro 'atomic_store_explicit'
  atomic_store_explicit(&__object->__flag, 0, __order);
  ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:394:2: error: incompatible type for argument 1 of '__atomic_store_n'
LD build/bootloader/bootloader.elf
CC build/esp32/intr_alloc.o
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_install':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:174:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->ref_count, 1);
     ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_uninstall':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:283:27: error: request for member '__val' in something not a structure or union
  __atomic_load_n(&(object)->__val, order)
                           ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:27:15: note: in definition of macro 'ETH_CHECK'
         if (!(a))                                                                 \
               ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:365:2: note: in expansion of macro 'atomic_load_explicit'
  atomic_load_explicit(object, memory_order_seq_cst)
  ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:215:15: note: in expansion of macro 'atomic_load'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
               ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:215:5: error: incompatible type for argument 1 of '__atomic_load_n'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
     ^
In file included from /Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_increase_reference':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:273:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_add(&(object)->__val, operand, order)
                              ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:355:2: note: in expansion of macro 'atomic_fetch_add_explicit'
  atomic_fetch_add_explicit(object, operand, memory_order_seq_cst)
  ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:341:5: note: in expansion of macro 'atomic_fetch_add'
     atomic_fetch_add(&eth_driver->ref_count, 1);
     ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:341:5: error: incompatible type for argument 1 of '__atomic_fetch_add'
In file included from /Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_decrease_reference':
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:279:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_sub(&(object)->__val, operand, order)
                              ^
/Users/skor/esp/esp-idf/components/newlib/include/stdatomic.h:361:2: note: in expansion of macro 'atomic_fetch_sub_explicit'
  atomic_fetch_sub_explicit(object, operand, memory_order_seq_cst)
  ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: note: in expansion of macro 'atomic_fetch_sub'
     atomic_fetch_sub(&eth_driver->ref_count, 1);
     ^
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/Users/skor/esp/esp-idf/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
make[1]: *** [src/esp_eth.o] Error 1
make: *** [component-esp_eth-build] Error 2
make: *** Waiting for unfinished jobs....
CC build/driver/sdio_slave.o

````

What can I do to solve this problem?

suda-morris, 2020-02-19:
please check your toolchain version `xtensa-esp32-elf-gcc -v `.

Since 4.0 ESP-IDF has switched to gcc8.2

See also https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-set-up-tools

korstiaanS, 2020-02-19:
Ok, but if I change, will the version 3.3 also work?
If not, how to go back?
I want to test with 4.0 but my live environment is all on 3.3 for the moment.

igrr, 2020-02-20:
If you follow the getting started guide for IDF v4.0, the required tools will not be installed into system PATH by default, so they won't interfere with other IDF versions. You should be able to go back to IDF v3.3 after you install v4.0.

Alvin1Zhang, 2020-03-12:
@korstiaanS Thanks for reporting, has your issue been fixed? Thanks.

korstiaanS, 2020-03-12:
Yes. I will close.

closed 2020-03-12 by korstiaanS

curlyz, 2021-02-26:
For future Googler.
On Ubuntu 18.04 if your version of gcc is not 8.4 do following :
Source : [https://askubuntu.com/questions/1028601/install-gcc-8-only-on-ubuntu-18-04](https://askubuntu.com/questions/1028601/install-gcc-8-only-on-ubuntu-18-04)
``` console
apt-get install -y software-properties-common 
add-apt-repository -y  ppa:ubuntu-toolchain-r/test
apt-get update
apt-get install -y gcc-8 g++-8
update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 700 --slave /usr/bin/g++ g++ /usr/bin/g++-7
update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 800 --slave /usr/bin/g++ g++ /usr/bin/g++-8
```