espressif/esp-idf #4805: DMA implementation for WS2812 LEDs instead of RMT to stop flickering

Francois-Belanger, 2020-02-19:
Using the RMT causes flickering of the led strip when using WiFi etc..  It seems the RMT is interrupt driven and NOT DMA.  Interrupt latency is an issue.

I'm proposing this solution using the great SPI DMA driver.  I've lost a lot of time trying to solve the issues of flickering with Task Priority, Core selection etc etc...

With this solution, one can drive up to 1000 color pixel or about 334 RGB Led.  I'm using the SPI because we don't need to re-copy every bytes to another buffer.  SPI DMA is using the provided user buffer, with I2S there is a massive copy going on.

`
#define LED_DMA_BUFFER_SIZE (LED_MAX_NBER_LEDS * sizeof(uint16_t) * 2 + 50)

typedef struct {
    spi_host_device_t host;
    spi_device_handle_t spi;
    int dma_chan;
    spi_device_interface_config_t devcfg;
    spi_bus_config_t buscfg;
} SPI_settings_t;

static uint16_t* ledDMAbuffer;

static SPI_settings_t SPI_settings = {
    .host = HSPI_HOST,
    .dma_chan = 2,
    .buscfg = {
        .miso_io_num = -1,
        .mosi_io_num = GPIO_NUM_2,
        .sclk_io_num = -1,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = LED_DMA_BUFFER_SIZE },
    .devcfg = { .clock_speed_hz = 3.2 * 1000 * 1000, //Clock out at 3.2 MHz
        .mode = 0, //SPI mode 0
        .spics_io_num = -1, //CS pin
        .queue_size = 1, //Not sure if needed
        .command_bits = 0,
        .address_bits = 0 }
};

### In the task: 
    esp_err_t err;

    err = spi_bus_initialize(SPI_settings.host, &SPI_settings.buscfg, SPI_settings.dma_chan);
    ESP_ERROR_CHECK(err);

    //Attach the Accel to the SPI bus
    err = spi_bus_add_device(SPI_settings.host, &SPI_settings.devcfg, &SPI_settings.spi);
    ESP_ERROR_CHECK(err);

    ledDMAbuffer = heap_caps_malloc(LED_DMA_BUFFER_SIZE, MALLOC_CAP_DMA); // Critical to be DMA memory.

### Update the strip
static void led_strip_update()
{
    uint16_t LedBitPattern[16] = {
        0x8888,
        0x888C,
        0x88C8,
        0x88CC,
        0x8C88,
        0x8C8C,
        0x8CC8,
        0x8CCC,
        0xC888,
        0xC88C,
        0xC8C8,
        0xC8CC,
        0xCC88,
        0xCC8C,
        0xCCC8,
        0xCCCC
    };
    uint32_t i, j, k;
    size_t data_written = 0;
    esp_err_t err;

    memset(ledDMAbuffer, 0, LED_DMA_BUFFER_SIZE);
    k = 0;
    for (i = 0; i < number_leds; i++) {
        uint32_t temp = table[i];// Data you want to write to each LEDs, I'm my case it's 95 RGB x 3 color
        ledDMAbuffer[k++] = LedBitPattern[0x0f & (temp >> 4)];
        ledDMAbuffer[k++] = LedBitPattern[0x0f & (temp)];
    }

    spi_transaction_t t;
    memset(&t, 0, sizeof(t));
    t.length = LED_DMA_BUFFER_SIZE * 8; //length is in bits
    t.tx_buffer = ledDMAbuffer;

    err = spi_device_transmit(SPI_settings.spi, &t);
    ESP_ERROR_CHECK(err);
}
`

Hope this help someone to enjoy this great platform :-)

koobest, 2020-02-19:
Hi,@Francois-Belanger

Please try the solution below：

1. Enable the following two items in menuconfig(->Component config ->Driver configurations ->SPI configuration):

[x] Place transmitting functions of SPI master into IRAM
[x] Place SPI master ISR function into IRAM

2. Put all SPI-related API calls on Core 1, including SPI initialization (spi_bus_initialize  spi_bus_add_device ...)

thanks !!

Francois-Belanger, 2020-02-20:
Hi, why is that ?  I'm proposing to use the SPI instead of the RMT ?  SPI runs fine :-). No ISR really needed with DMA.

negativekelvin, 2020-02-20:
Great but this doesn't really belong here since there is no driver maintained here.  Why don't you add it to the fastled library or as an example in esp-iot-solution?

closed 2020-02-21 by Francois-Belanger

Francois-Belanger, 2020-03-15:
Here is the fixed Array... fix endianness...
uint16_t LedBitPattern[16] = {
        0x8888,
        0x8C88,
        0xC888,
        0xCC88,
        0x888C,
        0x8C8C,
        0xC88C,
        0xCC8C,
        0x88C8,
        0x8CC8,
        0xC8C8,
        0xCCC8,
        0x88CC,
        0x8CCC,
        0xC8CC,
        0xCCCC
    };