espressif/esp-idf #4806: esp-tls-mbedtls: mbedtls_ssl_handshake returned -30592

chuledeco, 2020-02-19:
I'm trying to run the https_server example in an ESP32-PICO-D4.
I'm ussing: ESP-IDF:          v4.2-dev-318-g605da33c3-dirty

I also have an external SPI RAM:

I (483) spiram: Found 64MBit SPI RAM device
I (486) spiram: SPI RAM mode: flash 40m sram 40m
I (492) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1428) spiram: SPI SRAM memory test OK
I (1428) heap_init: Initializing. RAM available for dynamic allocation:
I (1429) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1435) heap_init: At 3FFB9400 len 00026C00 (155 KiB): DRAM
I (1441) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1447) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1454) heap_init: At 40099F5C len 000060A4 (24 KiB): IRAM
I (1465) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1508) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations

The server seems to be initialized correctly:

I (5568) example: Starting server
I (5568) esp_https_server: Starting server
I (5568) esp_https_server: Server listening on port 443
I (5578) example: Registering URI handlers
I (5u78) esp_netif_handlers: sta ip: 192.168.0.245, mask: 255.255.255.0, gw: 192.168.0.1
I (5578) example_connect: Got IP event!
I (5588) example_connect: Connected to juanamarea
I (5588) example_connect: IPv4 address: 192.168.0.245

But when I write http://192.168.0.245/ in the web browser I get rejected connection and the text bellow appears in the terminal:

E (986228) esp-tls-mbedtls: mbedtls_ssl_handshake returned -30592
E (986228) esp_https_server: esp_tls_create_server_session failed
W (986228) httpd: httpd_accept_conn: session creation failed
W (986238) httpd: httpd_server: error accepting new connection

Regards, Julian

negativekelvin, 2020-02-20:
https://github.com/espressif/esp-idf/issues/3624

mahavirj, 2020-02-24:
@chuledeco What is toolchain version you are using? You can find that using command `xtensa-esp32-elf-gcc --version`.

mahavirj, 2020-02-27:
@chuledeco Do you have any update?

mahavirj, 2020-03-02:
Closing due to lack of response. Please feel to re-open if issue still exists.

closed 2020-03-02 by mahavirj

saransarval, 2021-02-28:
Am also getting same issue on the hitting IP from browser.

saransarval, 2021-02-28:
[image]

As per the image am using  xtensa esp32 elf version is 8.2.0. please help on this