espressif/esp-idf #4807: How to transmit data through Uart with a range of above 300 bytes to 1024 bytes through uart in esp-idf 3.3

manojsai17, 2020-02-19:
I am unable to transfer data above 120bytes through uart through esp-idf using esp-wroom 32 module .

please suggest a solution or any code reference to send large data through uart using esp

koobest, 2020-02-23:
Hi, @manojsai17 
`int uart_write_bytes(uart_port_t uart_num, const char* src, size_t size)`  can send the data of the specified byte. Under what circumstances can you not send more than 120 bytes of data? Can you provide your test code?

thanks !!

koobest, 2020-02-24:
Hi,
This issue is the same as the other two:
https://github.com/espressif/esp-idf/issues/4820
https://github.com/espressif/esp-idf/issues/4719
please close this one, we will reply you here: https://github.com/espressif/esp-idf/issues/4820

thanks !!

Alvin1Zhang, 2020-03-02:
@koobest Thanks for notes. @manojsai17 Will close this one. Thanks.

closed 2020-03-02 by Alvin1Zhang