espressif/esp-idf #4809: Wifi Example - wifi station: connect to the AP fail

armandr, 2020-02-19:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:   
- Kit version: 
- Module or chip used: ESP-WROOM-32D Module
- IDF version (run ``git describe --tags`` to find it): v4.0 release
- Build System: CMake | IDF.PY
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows10
- (Windows only) environment type: ESP Command Prompt/Plain Command Prompt
- Using an IDE?: No
- Power Supply:       USB

## Problem Description

Wifi Example (examples\wifi\getting_started\station) fails to connect to AP (with correct credentials).  **This problem may not be limited to the example.**

"connect to the AP fail"

//Detailed problem description goes here.
The compilation goes fine, and the unit powers up, but it does not connect to the access point even though it has the right SSID and PWD.

I first noticed this after upgrading our project from V3.3 which worked fine to V4.0 and the device was no longer able to connect to our access point (same credentials stored in NVS).  

Even after flash erase, and ble_provisioning, it still failed to connect.

To rule out anything from our side, I configured SSID and PWD on the clean example, compiled and flashed with the following outcome:

### Expected Behavior
Connect to AP.
### Actual Behavior
"connect to the AP fail"
### Steps to reproduce

1. configure ESP_WIFI_PASSWORD, and ESP_WIFI_SSID to your AP
2. compile and flash

### Code to reproduce this issue
in examples\wifi\getting_started\station\main

```

## Debug Logs

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6876
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (70) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.0-dirty 2nd stage bootloader
I (40) boot: compile time 10:50:06
I (40) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 2MB
I (57) boot: Partition Table:
[... 757 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
```

zhangyanjiaoesp, 2020-02-20:
@armandr I have tested with v4.0, and I can connect to AP successfully. Maybe you can provide the  encryption method of your AP, and the type of your AP.

armandr, 2020-02-20:
@zhangyanjiaoesp thanks for verifying.  The issue was on the AP firmware.

closed 2020-02-20 by armandr

zhangyanjiaoesp, 2022-07-18:
> > @armandr I have tested with v4.0, and I can connect to AP successfully. Maybe you can provide the encryption method of your AP, and the type of your AP.
> 
> what do you mean by type of AP?

the product model