espressif/esp-idf #4810: Bidirectional 3-wire SPI mode

mirko, 2020-02-19:
**Is your feature request related to a problem? Please describe.**

I need to use the SPI 3-wire mode in a setup where not only 3 wires are used, but bidirectional communication is happening. So, technically, the same pin is used for MISO and MOSI.

I figured I can achieve that with the SPI 3 wire mode, however when calling the SPI related read() call, the MOSI pin is pulled high and no data comes in (seen and measured directly on the wire).
Deiniting SPI and clocking the SCK pin manually I see data coming from my target / SPI slave.

**Describe the solution you'd like**

I'd like to be able to use the SPI 3 wire mode for bidirectional communication - obviously half-duplex - on the same pin.

**Describe alternatives you've considered**

Using a hardware resistor hack as described in https://github.com/arduino/OpenOCD/blob/master/tcl/interface/ftdi/swd-resistor-hack.cfg was considered, however in my setup on ESP32 side I don't have any spare GPIO left.

Also re-init-ing the SPI stack for every turnaround (where the reading and writing side change swap) was considered - setting the previously used pin for mosi now for miso) - however the init process takes time and creates glitches on the wire (pulling up/down SCK/MISO/MOSI).

The API wouldn't change here - only data could be written *and* read to/from the same pin in 3-wire mode.

ginkgm, 2020-02-24:
Hi @mirko ,
1. Please initialize the SPI bus and leave the miso field `-1` (not used).
2. Initialize a new SPI device with `SPI_DEVICE_HALFDUPLEX` and  [`SPI_DEVICE_3WIRE`](https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#c.SPI_DEVICE_3WIRE) flags on the bus.

   Also please configure proper command, address, and dummy phase length. During the command and address phases, the MOSI is output, while during the dummy phase, the master is not reading neither.

BTW, I think this advanced feature is only provided in IDF, not in some simpified SDKs (like the Arduino).

loboris, 2020-03-01:
@ginkgm
this works, but only **if no DMA is used**.

I've tried to use `SPI_DEVICE_3WIRE` without `SPI_DEVICE_HALFDUPLEX` in DMA mode and it looks the `mosi` is correctly switched to input, but I've got some strange results:
- on first transaction (after setup) I correctly get the **first 8 bytes** (it looks **only 8 bytes are read**, the content of my rx buffer after the 8th byte is unchanged)
- on all following transactions **only 3 bytes are read** and they are all **0**

It is quite strange, as if the DMA is not configured correctly if **`SPI_DEVICE_3WIRE`** is used.
In my tests only the read phase was active.

As I need to read more than 64 bytes, I must use DMA, is there any solution to this issue?

Is there any way to transfer more than 64 bytes without using DMA (in the same transaction)? It looks it is not implemented (why? it shouldn't be hard to repeat 64 byte reads multiple times).

mirko, 2020-03-02:
@loboris This is documented as a known issue[1] where it says:

>  1. Half-duplex transactions are not compatible with DMA when both writing and reading phases are used.
> 
>         If such transactions are required, you have to use one of the alternative solutions:
> 
>         1. Use full-duplex transactions instead.
>         2. Disable DMA by setting the bus initialization function's last parameter to 0 as follows:
>             ``ret=spi_bus_initialize(VSPI_HOST, &buscfg, 0);

It basically says if you need to read *and* write (=bidirectional 3wire SPI) - which effectively *needs* to be half-duplex - you can not use DMA.

But I'm also really interested in the *why*.

[1] https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#known-issues

ginkgm, 2020-04-08:
Hi @loboris @mirko ,
It's determined by chip design, where we didn't expect the DMA will be used with half duplex mode.

Though you may use some tricks (for example, SW-controlled CS line with multiple transactions, all of which are in only one direction) to make it work, the throughput will be not as fast as DMA is used.

loboris, 2020-04-08:
@ginkgm 

I've solved this and I have **half-duplex 3-wire** transmission working great.
The throughput is only slightly lower than wnen using DMA, **2.66 us** pause after each 64 bytes (to write/read the spi buffer).
Tested at 26 MHz clock with transfers from 1 B to 64 KB, no issues.

No changes to spi_master are needed, I've only added one function to `spi_master.c`
```
/**
 * @brief Send a non-dma transaction, wait for it to complete, and return the result
 *
 * This function handles transmission of any length of data using a direct spi mode, without DMA
 *
 * @note This function is not thread safe when multiple tasks access the same SPI device.
 *      Normally a device cannot start (queue) polling and interrupt
 *      transactions simutanuously.
 *
 * @param handle Device handle obtained using spi_host_add_dev
 * @param trans_desc Description of transaction to execute
 * @param delay Delay in us between mosi and miso phase (if both are used
 *              if negative value is given, CS is deactivated before delay and activated after delay
 * @return
 *         - ESP_ERR_INVALID_ARG   if parameter is invalid
 *         - ESP_OK                on success
 */
esp_err_t spi_device_nodma_transmit(spi_device_handle_t handle, spi_transaction_t *trans_desc, int16_t delay);
```

I could make a PR if there is any interest for it.

ginkgm, 2020-04-09:
Hi @loboris ,
We already have a set of "nodma" transaction functions, see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#polling-transactions

But your PR will also be valuable for us to know what your really need and how you solve it.
And BTW, our driver design idea is not covering too much HW mechanisms, unless it's obviously a HW bug, or user cannot accomplish some reasonable applications above our driver.

windowsair, 2021-03-10:
Hi @loboris ,
I have encountered the same situation as you. In each half-duplex transfer transaction, no DMA is used and the length of the data involved does not exceed 64bits. In addition, I use a command phase instead of a write phase and include a read phase:

1.  First transmission, everything is fine.
2. For the second time, during the read phase, I noticed that the MOSI was always pulled up, which made the readout all 1s.

This issue arise with the 3-wire spi. To solve this, I had to externally connect the MOSI and MISO pins directly.

Any ideas? Thanks.😀

windowsair, 2021-03-10:
Ohhhhhh, I did it.😄

This is some of my experience in using 3-wire SPI, I hope it will be helpful to you:

1.  If the read phase is required in a transfer transaction (via MOSI), you need to set `sio` in the spi register to `1`.
2.  If you only need to perform write operations which include one or more of the command, address and write phase,  you need to clear `sio` to `0`

If you don't do this, there may be timing issues...

The above description is done at the register level. For esp-idf, you need to set or clear the `SPI_DEVICE_3WIRE` flag at the appropriate time.

Well, I haven't found other esp-idf examples yet, but it's really easy to misuse for a beginner like me.