espressif/esp-idf #4811: Nimble task crash after disconnect under heavy load.

Francois-Belanger, 2020-02-20:
v4.1-dev-2224-gbcd2358eb
CMake
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Mac OS

## Problem Description

After a series of connect/disconnect under heavy load, the BLE Nimble call the sync callback then the reset callback.

What is the required steps to fallback from a stack state reset ?  Maybe I have to redo the init ?

## Debug Logs

```
I (65422) BLE: BLE_GAP_EVENT_NOTIFY_TX, STATUS: 0
GATT procedure initiated: notify; 
att_handle=12
I (65547) BLE: BLE_GAP_EVENT_NOTIFY_TX, STATUS: 0
GATT procedure initiated: notify; 
att_handle=12
I (65548) BLE: BLE_GAP_EVENT_NOTIFY_TX, STATUS: 0
I (65809) BLE: disconnect; reason=531
---> Doing a ble_advertise();
GAP procedure initiated: advertise; 
disc_mode=2
adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0

Resetting state; reason=12
I (65851) BLE: Device Address: 
I (65851) BLE: c4:4f:33:35:26:3f

... crash...
```

prasad-alatkar, 2020-03-06:
Hi @Francois-Belanger apologies for delayed response. Though I was able to reproduce the issue when you had reported, I could not get time to debug it further. As it happens, I am not currently able to see the issue. Can you confirm if you are able to see the issue on latest master branch? Is it possible for you to provide any simple application based on `examples/bluetooth/nimble/bleprph` or `blehr` to reproduce the issue?

Francois-Belanger, 2020-03-10:
Hi, well I moved to another part of the project.  I will try to re-test with latest release this week.
Thanks

prasad-alatkar, 2020-03-26:
Hi @Francois-Belanger Just FYI, we have internally identified and fixed the issue, will update you once the changes are merged.

> What is the required steps to fallback from a stack state reset ? Maybe I have to redo the init ?

You don't have to redo the init, host stack should take care of this (given controller has reset successfully as well). Upstream PR to address this issue in NimBLE host stack: [https://github.com/apache/mynewt-nimble/pull/777](https://github.com/apache/mynewt-nimble/pull/777)

prasad-alatkar, 2020-05-26:
Hi @Francois-Belanger commit: be8c847b fixes this issue, please let me know if you still face the issue.

closed 2020-05-26 by Alvin1Zhang