espressif/esp-idf #4812: How to use default mqtt esp library when gsm module is interfaced with esp through uart communication to communicate the cloud

manojsai17, 2020-02-20:
we interfaced the simcom 7000 gsm module to esp-wroom 32 ,
how to use default mqtt library when esp module is interfaced with sim 7000 e gsm module

umer-ilyas, 2020-02-24:
use the `pppos_client ` example, it also uses esp_mqtt

**examples\protocols\pppos_client**

manojsai17, 2020-02-24:
problem is that I used a traditional at commands method to connect to internet using sim 7000 gsm module .

david-cermak, 2020-02-28:
@manojsai17 Please start with this [example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client) as @umer-ilyas  suggested. It also uses `AT` commands before switching to the PPP mode. If you have any specific issue or question, please post it here.
In case you are already connected to the network (got an IP address from the GSM module), please look at some [mqtt example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/tcp) and just use everything after [example_connect()](https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c#L148) , so replacing the WiFi/Ethernet connection with your own GSM.

manojsai17, 2020-02-28:
how to link the ip address got from my gsm module to esp module . and use default  mqtt library .
I dont want to use ppos method . becuase i dont have knowledge exposure on ppos method .

david-cermak, 2020-02-28:
@manojsai17 I still don't quite understand. Could you please describe your setup in more detail? So you've already received an IP address from the GSM module and you can interface it from the ESP to talk to the internet over TCP, correct? Could you give an example of such interface?

manojsai17, 2020-02-28:
we interface the gsm module to esp uart .
I able to data transfer to cloud using wifi with default mqtt library .
whenever there is  no wifi connectivity , i have to use gsm module .
So we interfaced gsm module to esp uart .
with traditional at command i will get the network and ip address .
so how to link the ip of gsm module which is attached to esp32 .to transfer the data to cloud using mqtt

david-cermak, 2020-02-28:
okay, so if I understand correctly you can open a TCP/IP "socket" using the at commands and talk to remote servers, right? In that case it's much easier to use the pppos approach since it is already implemented and you can directly get to the lwip, the ESP network stack, as the mqtt library uses tcp_transport component connecting to it. It is possible however to add another "layer" in `tcp_transport` component to pass the TCP  traffic to custom interface (network by uart in your case) to bypass the ESP network stack. This might be a lot of work though.

manojsai17, 2020-02-28:
actually i dont have experience on pppos related concept . 
I written code upto , with at commands , where netwrok connecton , getting ip address .
so is there any way link the ip address , open the tcp port .and send the data using default mqtt library .

david-cermak, 2020-03-02:
Not really, If you open a tcp port/socket all traffic goes to the underlying TCP/IP stack and then the network data could de passed to any arbitrary interface (such as uart), but these are level 2 data, not only the tcp traffic.
AFAICT these three options are available to pass the mqtt data over uart as tcp only data:
* PPP client - as mentioned, off the shelf, supported and working solution
* Add a new implementation of [tcp layer](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_tcp.c#L52) of `tcp_transport` bypassing the lwip socket and direct the tcp traffic to arbitrary interface
*  Add a new `esp_netif` driver to direct the network data to an arbitrary interface, but also add a logic to strip the lower layers and pick only the tcp data.

diogoviannaaraujo, 2020-05-20:
Ive been trying to use the PPPoS client example with SIM7000 let-m modem, there are differences in the initial AT commands and there are some random crashing that causes it to connects and disconnect right after.

Also was not able to reconnect after the connection was lost, could not find much example/documentation about how we should handle connection loss and stuff..

But I was able to occasionally connect, get IP and publish to the MQTT server with it, the overall process is the same to the SIM800.

After I get it working ok (I have 2 week for that 😭) is desirable to have more modems configurations in the example and I should make a pull request? Is there a place to share this kind of stuff? Im new to this embedded ecosystem thing

Anyway, if anyone is struggling with the same issues and want to share experiences get in touch, maybe I can pay you a beer and we talk about making some nice modem support on idf.

david-cermak, 2020-05-20:
@diogoviannaaraujo Yes, please. if you're willing to share your implementation, please file a PR. would really appreciate to have another modem supported, thanks for your help!

Alvin1Zhang, 2020-05-26:
@diogoviannaaraujo Thanks for reporting. Will close the issue now, feel free to reopen with more updates or file a PR which would be appreciated. Thanks.

closed 2020-05-26 by Alvin1Zhang