espressif/esp-idf #4813: pyparsing requirements not satisfied with Ubuntu 20.04

hb-kaertech, 2020-02-20:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-358-ga0644bf8a-dirty
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

On Ubuntu 20.04, the packages pyparsing version in 2.4.6 (https://packages.ubuntu.com/focal/python-pyparsing). Thus it does not satisfies the esp-idf requirements (`pyparsing>=2.0.3,<2.4.0`).

### Expected Behavior

`esp.py build` works correctly

### Actual Behavior

`esp.py build` fails with the following error:

> The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide

### Steps to reproduce

1. Install Ubuntu 20.04
2. Try to compile any example

## Fix

Can you update esp-idf so it works with pyparsing 2.4?

dobairoland, 2020-02-20:
> Can you update esp-idf so it works with pyparsing 2.4?

Hi @hb-kaertech. Unfortunately we cannot at the moment because there is an outstanding compatibility issue which was promised to get fixed in 2.5 sometime in the future.

However, if you use the `install.sh` and `setup.sh` scripts then it should install and use the right package inside a virtual Python environment. https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id3, https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id5

hb-kaertech, 2020-02-21:
Indeed, I did think about re-running install.sh after the upgrade to 20.04. Thank you.

For other people in the future, here is the commit discussing the incompatibility: 08adc39102d7defd876118a779432bbd31866c5f#diff-b4ef698db8ca845e5845c4618278f29a

closed 2020-02-21 by hb-kaertech