espressif/esp-idf #4814: Shift 32 bits variable in ULP Coprocessor

Matheusmbgm, 2020-02-20:
Hello guys,

I need to interface HX711 with ESP32 in deep sleep mode. That shouldn't be a problem because interfacing with HX711 is really simple, just putting some pins up, some pins down and read some pins and do some basic math.

The biggest challenge i'm facing right now is to work with 32 bits variable. At some point of the code i need to shift 1 bit left, and thats really hard to do because all 4 registers are 16 bits only. I'm having trouble right now at the line "jump b_lsb_1, 0x8000". Compiler gives me this error : "Jump adress is to far. Input text was eq.".

Code is below and i haven't finished yet, i'm trying to do part by part. If someone could please give me some advice of a easiest way to work with 32 bits variables on ULP or even a easiest way to interface HX711 with ULP ESP32, i would be very pleased.
PS: Sorry 4 PT-BR comments in the code :lol: , the parts that i have doubts are commented in english :D
Thank u very much,
Matheus

#include "soc/soc_ulp.h"
#include "soc/rtc_io_reg.h"
#include "soc/sens_reg.h"
#include "soc/rtc_cntl_reg.h"

.bss//Variaveis sao declaradas dentro da secao .bss

.global measurement_count
measurement_count: .long 0

.text//O codigo é feito dentro da secao .text

.global main
main://O codigo e iniciado aqui, equivale ao void setup()

```cpp
jump loop
loop: //add0 = 9 adsk = 8
WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG, RTC_GPIO_OUT_DATA_W1TS_S+9, 1, 1)//sobe add0
WRITE_RTC_REG(RTC_GPIO_OUT_W1TC_REG, RTC_GPIO_OUT_DATA_W1TC_S+8, 1, 1)//desce adsk
le_add0:
READ_RTC_REG(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT_S + 9, 1) // le gpio_num_15 = rtc gpio 13
and r0, r0, 1 // faz uma and do valor lido com 0x01
jumpr le_add0, 0, eq // se o valor de r0 for igual zero, volta para ler dnv r0
stage_rst
loop_count:
WRITE_RTC_REG(RTC_GPIO_OUT_W1TS_REG, RTC_GPIO_OUT_DATA_W1TC_S+8, 1, 1)//desce adsk
and r0, r1, 0x8000 // R0 = R1 & 0X8000
jumpr b_lsb_1, 0x8000, // if R1 (LSB BYte) MSB BIT is 1, R0 will be equal to 0x8000 then jump to b_lsb_1
jump b_lsb_0 // if R1(LSB Byte) MSB bit is 0, R0 will be 0x0000
b_lsb_0: // R1 MSB Bit equal 0
lsh r0, r1, 1 // shift 1 bit left from LSB Byte
mov r1, r0
lsh r0, r2, 1 // shift 1 bit left from R2( MSB Byte)
mov r2, r0
jump loop_count
b_lsb_1: // R1 MSB Bit equal 0
lsh r0, r1, 0x01 // shift 1 bit left from lsb byte
mov r1, r0
lsh r0, r2, 0x01 // shift 1 bit left from msb byte
add r2, r0, 1 // add 1 to the value that correspondsto MSB bit from LSB byte that now is LSB bit of MSB Byte.
jump loop_count
```

stickbreaker, 2020-02-20:
Shouldn't
```
jumpr b_lsb_1, 0x8000, // if R1 (LSB BYte) MSB BIT is 1, R0 will be equal to 0x8000 then jump to b_lsb_1
```
be:
```
jumpr b_lsb_1, 0x8000, GE // if R1 (LSB BYte) MSB BIT is 1, R0 will be equal to 0x8000 then jump to b_lsb_1
```

Matheusmbgm, 2020-02-21:
Yes, it is, i forgot to put it in the code, but did you got my logic? Is there any easiest way?

Matheusmbgm, 2020-02-21:
Another question, i'm searching but what is the easiest way to read one 32bits variable of my ulp program in my main program?

stickbreaker, 2020-02-21:
The ULP hardware can only access the lower 16bits of memory. It is styled as a 16bit cpu.  So, no.  You have to work with 16bit values. to pass a 32 bit value to the main extensa cpu's you have to use combine 2 32bit memory locations:
```c++
// c code
extern long ulp_var1;
extern long ulp_var2;

// asm code
    .global var1
var1:  .long 0
    .global var2
var2:  .long 0

// c code
uint32_t measurement = ((var2 & 0xffff) << 16) | (var1 & 0xffff);
```
Your asm code is incomplete, i don't have any advice because I don't know what it is trying to do.

Chuck.

alotionIG, 2022-07-21:
Hi, @Matheusmbgm , I just want to do a confirmation, did you post your desired code, here? 
https://www.esp32.com/viewtopic.php?t=14349

thanks you