espressif/esp-idf #4816: CAN driver TX queue size error

obdevel, 2020-02-21:
The CAN driver TX queue length seems to be set incorrectly.

e.g. I set the TX queue length to 5:

g_config.tx_queue_len = 5;

But if the bus is disconnected and I continue to send frames with can_transmit(), it does not error (with ESP_ERR_TIMEOUT) until there are 6 frames in the queue, as measured with can_get_status_info() and the .msgs_to_tx member of the can_status_info_t type.

A minor bug.

Dazza0, 2020-02-21:
@obdevel Not a bug, it's by design.
First message skips the TX queue and is put directly into the TX buffer. The message remains in the TX buffer indefinitely because the bus is disconnected. The second to sixth message are put into the queue because the TX buffer is occupied by the first message.

obdevel, 2020-02-21:
Thanks for the explanation !

closed 2020-02-21 by obdevel