espressif/esp-idf #4817: Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

AxelLin, 2020-02-21:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-414-g132cc67c0336
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

Hi,
I sometimes got below panic in my application.

The /home/axel/esp/esp-idf-dev/components/third_party/button actually is copied from
ESP-IDF examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client/components/button.
Any idea about this issue?

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40081d96  PS      : 0x00060034  A0      : 0x80081e45  A1      : 0x3ffb05f0
0x40081d96: timer_insert at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer.c:213

A2      : 0x3ffbbf28  A3      : 0x00000000  A4      : 0x3ff60000  A5      : 0x3ffb3b50
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffbbf28  A9      : 0x3ffb3c34
A10     : 0x0233c322  A11     : 0x00000000  A12     : 0x3ffbbf28  A13     : 0x3ffbdab0
A14     : 0x00000000  A15     : 0x3ffb3cc0  SAR     : 0x00000002  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40081f4d  LEND    : 0x40081f55  LCOUNT  : 0x00000027
0x40081f4d: esp_timer_impl_get_counter_reg at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:126 (discriminator 2)

0x40081f55: esp_timer_impl_get_counter_reg at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:136 (discriminator 1)

Core 0 was running in ISR context:
EPC1    : 0x40198dea  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081d96
0x40198dea: esp_pm_impl_waiti at /home/axel/esp/esp-idf/components/esp32/pm_esp32.c:486

0x40081d96: timer_insert at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer.c:213

ELF file SHA256: ba240a31925293be0d5ecaa69b58df02e9f40f99c25f4b93fe99edac8d913030

Backtrace: 0x40081d93:0x3ffb05f0 0x40081e42:0x3ffb0610 0x400e049c:0x3ffb0630 0x40084099:0x3ffb0650 0x400840e6:0x3ffb0670 0x40082469:0x3ffb0690 0x40198de7:0x3ffbda70 0x400d4845:0x3ffbda90 0x40093918:0x3ffbdab0
0x40081d93: timer_insert at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer.c:213

ELF file SHA256: ba240a31925293be0d5ecaa69b58df02e9f40f99c25f4b93fe99edac8d913030

Backtrace: 0x40081d93:0x3ffb05f0 0x40081e42:0x3ffb0610 0x400e049c:0x3ffb0630 0x40084099:0x3ffb0650 0x400840e6:0x3ffb0670 0x40082469:0x3ffb0690 0x40198de7:0x3ffbda70 0x400d4845:0x3ffbda90 0x40093918:0x3ffbdab0
0x40081d93: timer_insert at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer.c:213

0x40081e42: esp_timer_start_once at /home/axel/esp/esp-idf/components/esp_timer/src/esp_timer.c:138

0x400e049c: button_gpio_isr_handler at /home/axel/esp/esp-idf-dev/components/third_party/button/button.c:230

0x40084099: gpio_isr_loop at /home/axel/esp/esp-idf/components/driver/gpio.c:405

0x400840e6: gpio_intr_service at /home/axel/esp/esp-idf/components/driver/gpio.c:431

0x40082469: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1164

0x40198de7: esp_pm_impl_waiti at /home/axel/esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d4845: esp_vApplicationIdleHook at /home/axel/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40093918: prvIdleTask at /home/axel/esp/esp-idf/components/freertos/tasks.c:3385

Core 1 register dump:
PC      : 0x40198dea  PS      : 0x00060634  A0      : 0x800d4848  A1      : 0x3ffbe1e0
0x40198dea: esp_pm_impl_waiti at /home/axel/esp/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x00060023
A6      : 0x00000001  A7      : 0x00060723  A8      : 0x800d458e  A9      : 0x3ffbe1b0
A10     : 0x00000000  A11     : 0x3ffb3d38  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: ba240a31925293be0d5ecaa69b58df02e9f40f99c25f4b93fe99edac8d913030

Backtrace: 0x40198de7:0x3ffbe1e0 0x400d4845:0x3ffbe200 0x40093918:0x3ffbe220
0x40198de7: esp_pm_impl_waiti at /home/axel/esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d4845: esp_vApplicationIdleHook at /home/axel/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40093918: prvIdleTask at /home/axel/esp/esp-idf/components/freertos/tasks.c:3385

Campou, 2020-02-21:
Hi @AxelLin 

I have tried the button component under `ble_mesh_node/onoff_client`, it works fine. Also this button component is copied from https://github.com/espressif/esp-iot-solution/tree/master/components/general/button.

Thanks.

AxelLin, 2020-02-21:
Hi @Campou,
It does not always hit the panic, but happen occasionally.

BTW, below is how I use the button component:

static void button_reset_default_cb(void* arg)
{
    ESP_LOGI(TAG, "%s: %s", __func__, (char *)arg);
}

button_handle_t rst_btn_handle = iot_button_create(RESET_DEFAULT_BUTTON, RESET_DEFAULT_BUTTON_ACTIVE_LEVEL);
if (rst_btn_handle) {
        /* Long press for 3 seconds to reset default */
        iot_button_add_custom_cb(rst_btn_handle, 3, button_reset_default_cb, "ResetDefault");
}

AxelLin, 2020-02-24:
This issue only happen when BUTTON_USE_ESP_TIMER=y
If I choose BUTTON_USE_RTOS_TIMER=y instead, this issue never happen.

igrr, 2020-02-24:
@AxelLin Could you please try setting `CONFIG_ESP_TIMER_IMPL_FRC2=y` in sdkconfig, to use the older esp_timer implementation? This way we should know if this is a bug in esp_timer or elsewhere.

AxelLin, 2020-02-24:
Hi @igrr ,

It still happen if setting CONFIG_ESP_TIMER_IMPL_FRC2=y
It seems this panic is not 100% related to pressing the button.
Sometimes it happen even I didn't touch the button.

Below shows the log.
PS. I already comment out most of my application code.
In my code, GPIO[34] is for the button.

entry 0x40080684
I (30) boot: ESP-IDF v4.2-dev-430-g7092613eea29-dirt 2nd stage bootloader
I (30) boot: compile time 09:43:51
I (32) boot: chip revision: 1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (80) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (88) boot:  3 test             test app         00 20 00020000 000e0000
I (95) boot:  4 ota_0            OTA app          00 10 00100000 00140000
I (103) boot:  5 ota_1            OTA app          00 11 00240000 00140000
I (110) boot:  6 res              Unknown data     01 82 00380000 00040000
I (118) boot:  7 stage            Unknown data     01 82 003c0000 00040000
I (125) boot: End of partition table
I (130) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x07f44 ( 32580) map
I (151) esp_image: segment 1: paddr=0x00107f6c vaddr=0x3ffb0000 size=0x020f0 (  8432) load
I (155) esp_image: segment 2: paddr=0x0010a064 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/axel/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1789

I (159) esp_image: segment 3: paddr=0x0010a470 vaddr=0x40080404 size=0x05ba8 ( 23464) load
I (178) esp_image: segment 4: paddr=0x00110020 vaddr=0x400d0020 size=0x18074 ( 98420) map
0x400d0020: _stext at ??:?

I (216) esp_image: segment 5: paddr=0x0012809c vaddr=0x40085fac size=0x03260 ( 12896) load
0x40085fac: rtc_clk_32k_enable_common at /home/axel/esp/esp-idf/components/soc/src/esp32/rtc_clk.c:126

I (227) boot: Loaded app from partition at offset 0x100000
I (227) boot: Disabling RNG early entropy source...
I (228) cpu_start: Pro cpu up.
I (231) cpu_start: Application information:
I (236) cpu_start: Project name:     app-fw
I (241) cpu_start: App version:      0.1.1.17
I (246) cpu_start: Compile time:     Feb 24 2020 19:28:27
I (252) cpu_start: ELF file SHA256:  4a71002ea1f51622...
I (258) cpu_start: ESP-IDF:          v4.2-dev-430-g7092613eea29-dirt
I (265) cpu_start: Starting app cpu, entry point is 0x40081128
0x40081128: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp32/cpu_start.c:274

I (0) cpu_start: App cpu up.
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (282) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (288) heap_init: At 3FFB2930 len 0002D6D0 (181 KiB): DRAM
I (295) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (301) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (307) heap_init: At 4008920C len 00016DF4 (91 KiB): IRAM
I (314) cpu_start: Pro cpu start user code
I (332) spi_flash: detected chip: gd
I (332) spi_flash: flash io: dio
I (332) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (473) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (473) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (481) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (490) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (499) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 113 lines cut ...]
I (247) cpu_start: Compile time:     Feb 24 2020 19:28:27
I (253) cpu_start: ELF file SHA256:  4a71002ea1f51622...
I (259) cpu_start: ESP-IDF:          v4.2-dev-430-g7092613eea29-dirt
I (266) cpu_start: Starting app cpu, entry point is 0x40081128
0x40081128: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp32/cpu_start.c:274

I (257) cpu_start: App cpu up.
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (283) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (289) heap_init: At 3FFB2930 len 0002D6D0 (181 KiB): DRAM
I (295) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (302) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (308) heap_init: At 4008920C len 00016DF4 (91 KiB): IRAM
I (314) cpu_start: Pro cpu start user code
I (332) spi_flash: detected chip: gd
I (333) spi_flash: flash io: dio
I (333) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (473) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (473) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (481) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (490) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (499) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
use esp timer !!!
I (532) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3

igrr, 2020-02-24:
> Sometimes it happen even I didn't touch the button.

> 0x400d3849: button_gpio_isr_handler

This does indicate that the GPIO ISR has happened... Do you have an external pull-up resistor on the button GPIO? You do need that since GPIO34 doesn't have an internal pull-up. Also a small capacitor (0.1uF) from GPIO34 to ground can help as a debouncing measure.

AxelLin, 2020-02-25:
Hi @igrr,
I will ask our h/w guys when I back to office next week.
But I feel a debounce issue should not cause panic.
Anyway, will update to you after adding external pull-up resistor.

Thanks.

AxelLin, 2020-03-08:
Cannot reproduce the issue after adding external pull-up resistor.

closed 2020-03-08 by AxelLin