espressif/esp-idf #4818: ESP_GAP_SEARCH_INQ_CMPL_EVT sometimes does not fire

ftoyon-zbre, 2020-02-21:
## Environment
- Development Kit: ESP32-DevKitC and ESP32-Ethernet-Kit
- Kit version (for DevKitC): v4
- Kit version (for ESP32-Ethernet-Kit): A_V1.1
- Module or chip used:  ESP32-WROOM-32 and ESP32-WROVER
- IDF version: v3.3.1
- Build System: Make
- Compiler version: -1.22.0-80-g6c4433a5
- Operating System: Windows
- (Windows only) environment type: MSYS2 mingw32.
- Power Supply: USB or PoE

## Problem Description
We loop on one second scans and between two scans we process the registered advertinsings:
```
	esp_err_t errRc = esp_ble_gap_set_scan_params(&m_scan_params);
	errRc = esp_ble_gap_start_scanning(1);
```

In the treatment of the event `SP_GAP_SEARCH_INQ_RES_EVT` we check if the MAC is part of a list (max 10 items) if in the list we check if the advertising is already registered if it is not registered we added to the the list (max 10 advertising).

In the processing of the event `ESP_GAP_SEARCH_INQ_CMPL_EVT` we wake up our task of processing advertisings received.

Once the advertisings have been processed, we loop on start scan.

The problem is that in environments with a very large number of objects other than ours, at times the event `ESP_GAP_SEARCH_INQ_CMPL_EVT` never occurs.

In this case we force a scan stop:
```
	esp_err_t errRc = esp_ble_gap_stop_scanning();
```

Then we wait for the `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` event.

Many times this is enough to trigger the reception of the event `ESP_GAP_SEARCH_INQ_CMPL_EVT` followed by the event` ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT`.

But sometimes none of these events are triggered, but we continue to receive events `ESP_GAP_SEARCH_INQ_RES_EVT`.

In this case if we launch a scan it works, but sometimes we get stuck in the call to the function `esp_ble_gap_set_scan_params`.

gengyuchao, 2020-02-24:
Hello, do you processing the received advertisings in the `ESP_GAP_SEARCH_INQ_CMPL_EVT`? Or processing it in another task?

ftoyon-zbre, 2020-02-24:
Hi !
The processing is carried out in an independent task following the reception of the event `ESP_GAP_SEARCH_INQ_CMPL_EVT`.
This task waits for the `give` of a semaphore realized during the reception of the event or a timeout of 2s.
It is in the case of the triggering of the timeout that we try the `esp_ble_gap_stop_scanning`.

gengyuchao, 2020-02-24:
Do you print some log when got the scan result? If yes, can you try to set the timeout as a larger number like 5s? Just to confirm that if the event is late.

ftoyon-zbre, 2020-02-24:
Your idea of waiting a little longer seems the right track.
To validate it, we added a second timeout in order to keep the original timeouts.
And indeed we note on the few cases which appeared that the event `ESP_GAP_SEARCH_INQ_CMPL_EVT` is triggered shortly after the original timeout.

Do you have an idea of this delay of more than one secode in the arrival of this events and the maximum delay?

This large lag is a problem because we need to quickly connect to our objects that only advertise for 5 seconds.

We continue the tests and as soon as we have traces with a lockup we send it to you.

gengyuchao, 2020-02-25:
Since ESP_GAP_SEARCH_INQ_CMPL_EVT(and so on events) is issued by a callback function in the BTC queue, if you do some handling in the event (such as print), that will take some time and delay the later events.

ftoyon-zbre, 2020-02-25:
Is it possible to limit the number of callbacks due to the event `ESP_GAP_SEARCH_INQ_RES_EVT` which is called 99 times out of 100 for nothing?
I saw that there was a white list but the clear description of the mechanism or example.
Can this limit the number of callback for this event and reduce the delay?

gengyuchao, 2020-02-25:
For the first question. Sorry, I don't understand what you said.
For the second question.
Yes, there are two ways to reduce this delay. One is to put the scan result data into a queue and create a new task to process it. As you said, another way is to create a whitelist and only allow the controller to report the devices in the whitelist, which can greatly reduce the delay.
you can refer to the function 
```c
esp_err_tesp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_tremote_bda, esp_ble_wl_addr_type_twl_addr_type)
```
and set the scan_filter_policy member in esp_ble_scan_params_t as BLE_SCAN_FILTER_ALLOW_ONLY_WLST

```c
static esp_ble_scan_params_t ble_scan_params = {
    .scan_filter_policy     =  BLE_SCAN_FILTER_ALLOW_ONLY_WLST,
}
```

ftoyon-zbre, 2020-02-25:
Less than 1% of advertisings come from a monitored address. We would like to receive the event `ESP_GAP_SEARCH_INQ_RES_EVT` only for this one.

We trying to use the whitelist but the `ESP_GAP_SEARCH_INQ_RES_EVT` event is triggered even for addresses that are not in it.

```C
esp_err_tesp_ble_gap_update_whitelist(true, a_remote_bda);
/*...*/
m_scan_params.scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ONLY_WLST;
esp_ble_gap_set_scan_params(&m_scan_params);
```

As a reminder, we are using version 3.3.1.

ftoyon-zbre, 2020-02-26:
Yesterday we had a complete shutdown of the reception of BLE GAP events.
Traces are available [here](https://gist.github.com/ftoyon-zbre/c9f4b263e18310941028be8723877934)
We can see from line 652 that we have no more events.
And finally in line 1323 we call the function `esp_ble_gap_stop_scanning` without return.

P.S.: the 588 events `ESP_GAP_SEARCH_INQ_RES_EVT` of this trace are not useful to us.

ftoyon-zbre, 2020-02-26:
> Less than 1% of advertisings come from a monitored address. We would like to receive the event `ESP_GAP_SEARCH_INQ_RES_EVT` only for this one.
>
> We trying to use the whitelist but the [...]

Finally this works if at the same time you activate the filtering of duplicates.
```
	m_scan_params.scan_duplicate = BLE_SCAN_DUPLICATE_ENABLE;
```

gengyuchao, 2020-02-26:
Hello, did you open the duplicate option in the menuconfig at `→ Component config → Bluetooth → Bluetooth controller→ BLE Scan Duplicate Options`?

ftoyon-zbre, 2020-02-26:
My BLE SCAN sdkconfig
```
CONFIG_BLE_SCAN_DUPLICATE=y
CONFIG_SCAN_DUPLICATE_BY_DEVICE_ADDR=
CONFIG_SCAN_DUPLICATE_BY_ADV_DATA=
CONFIG_SCAN_DUPLICATE_BY_ADV_DATA_AND_DEVICE_ADDR=y
CONFIG_SCAN_DUPLICATE_TYPE=2
CONFIG_DUPLICATE_SCAN_CACHE_SIZE=100
```

GYC-Espressif, 2023-11-15:
Hello, this issue has been resolved after commit [c7a8025d].

closed 2023-11-15 by Alvin1Zhang