espressif/esp-idf #4819: WebSocket server component/example

fri-sch, 2020-02-22:
**Is your feature request related to a problem? Please describe.**

The framework includes a WebSocket client component/example, but no WebSocket server implementation. For many applications were the ESP32 acts as a host to other devices a WebSocket server is required.

**Describe the solution you'd like**

A WebSocket server component and example, based on the existing code and usable together with the existing HTTP/HTTPS server components.

**Describe alternatives you've considered**

I've been looking for other implementations and examples, but couldn't find anything that integrates with the idf and the HTTP server nicely.

huming2207, 2020-02-24:
You probably can try my fork, but still waiting to be merged to mainline: https://github.com/espressif/esp-idf/pull/4306

AshUK, 2020-02-29:
Would like to this supported and maintained as part of the ESP-IDF too +1

closed 2020-03-19 by espressif-bot