espressif/esp-idf #4821: esp-idf_tools-setup-2.2.exe fails with exit code 1

hrecently, 2020-02-22:
Hello,
today I did my second try to install the ESP-IDF but I failed again at Step 1 (Install the prerequisites for Windows):
- I choosed to do it with the installer from https://dl.espressif.com/dl/esp-idf-tools-setup-2.2.exe
- I started esp-idf_tools-setup-2.2.exe as admin
- Available Python version: Python 3.7 (64-bit)
- Available Git versions: Install Git 2.21.0
- Available ESP-IDF versions: Download ESP-IDF
- Download ESP-IDF: v4.0 (release version)
- Install:
 ... Downloading additional files (396,08MB)
 ... Extracting files ...
 ... Saving unistall information
 ... Finishing installation
 ... Extracting files ...
 ... Installing Git
 ... Extracting ESP-IDF
 ... Finishing ESP-IDF installation
 ... Installing ESP-IDF tools
 ... Running command ...Desktop\esp-idf\tools.py install
 ... raise CalledProcessError(retcode,cmd)
 ... Message Window Setup: Installation has failed with exit code 1

Here the last lines of the huge log:

2020-02-22 11:20:32.877   --------
2020-02-22 11:20:32.877   idf_tools.py exists in IDF directory
2020-02-22 11:20:32.877   idf_tools.py command: C:\Users\Bettina\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py
2020-02-22 11:20:32.877   Installing tools:C:\Users\Bettina\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py install
2020-02-22 11:22:09.325   Done, exit code=0
2020-02-22 11:22:09.325   --------
2020-02-22 11:22:09.325   Running command: C:\Users\Bettina\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py install
                          Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@esp-2019r2-8.2.0
                          file xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\Bettina\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0
                          Installing esp32ulp-elf@2.28.51.20170517
                          file binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\Users\Bettina\.espressif\tools\esp32ulp-elf\2.28.51.20170517
                          Installing cmake@3.13.4
                          file cmake-3.13.4-win64-x64.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\cmake-3.13.4-win64-x64.zip to C:\Users\Bettina\.espressif\tools\cmake\3.13.4
                          Installing openocd-esp32@v0.10.0-esp32-20190313
                          file openocd-esp32-win32-0.10.0-esp32-20190313.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip to C:\Users\Bettina\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313
                          Installing mconf@v4.6.0.0-idf-20190628
                          file mconf-v4.6.0.0-idf-20190628-win32.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to C:\Users\Bettina\.espressif\tools\mconf\v4.6.0.0-idf-20190628
                          Installing ninja@1.9.0
                          file ninja-1.9.0-win64.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\ninja-1.9.0-win64.zip to C:\Users\Bettina\.espressif\tools\ninja\1.9.0
                          Installing idf-exe@1.0.1
                          file idf-exe-v1.0.1.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\idf-exe-v1.0.1.zip to C:\Users\Bettina\.espressif\tools\idf-exe\1.0.1
                          Installing ccache@3.7
                          file ccache-3.7-w64.zip is already downloaded
                          Extracting C:\Users\Bettina\.espressif\dist\ccache-3.7-w64.zip to C:\Users\Bettina\.espressif\tools\ccache\3.7
                          
2020-02-22 11:22:09.325   --------
2020-02-22 11:22:09.325   Installing Python environment:C:\Users\Bettina\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py install-python-env
2020-02-22 11:22:10.060   Done, exit code=1
2020-02-22 11:22:10.060   --------
2020-02-22 11:22:10.060   Running command: C:\Users\Bettina\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py install-python-env
                          usage: virtualenv [--version] [--with-traceback] [-v | -q] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_separated_list] [--clear] [--system-site-packages]
                                            [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel] [--clear-app-data] [--symlink-app-data]
                                            [--prompt prompt] [-h]
                                            dest
                          virtualenv: error: unrecognized arguments: --no-site-packages
                          Creating a new Python environment in C:\Users\Bettina\.espressif\python_env\idf4.0_py3.7_env
                          Traceback (most recent call last):
                            File "C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py", line 1372, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py", line 1368, in main
                              action_func(args)
                            File "C:\Users\Bettina\Desktop\esp-idf\tools\idf_tools.py", line 1184, in action_install_python_env
                              stdout=sys.stdout, stderr=sys.stderr)
                            File "C:\Users\Bettina\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 347, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['C:\\Users\\Bettina\\AppData\\Local\\Programs\\Python\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\Bettina\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 2.
                          
2020-02-22 11:22:10.060   --------
2020-02-22 11:22:10.060   Message box (OK):
                          Installation has failed with exit code 1
2020-02-22 11:23:42.341   User chose OK.
2020-02-22 11:23:42.341   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-02-22 11:23:47.701   User chose Yes.
2020-02-22 11:23:47.747   CurStepChanged raised an exception.
2020-02-22 11:23:47.778   Need to restart Windows? No
2020-02-22 11:23:47.778   Exception message:
2020-02-22 11:23:47.778   Message box (OK):
                          Out Of Range.
2020-02-22 11:23:47.810   User chose OK.
2020-02-22 11:23:47.810   Deinitializing Setup.
2020-02-22 11:23:47.841   Log closed.

I need help ...
friendly regards
recently

krzychb, 2020-02-22:
Hi @hrecently, similar error message has been reported and resolved in https://github.com/espressif/esp-idf/issues/4744#issuecomment-586262996, please check it out.

hrecently, 2020-02-22:
Ok, Thank you. Next Version of the Installer will be released soon. I wait ...

igrr, 2020-02-24:
Unfortunately we can't update the docs at docs.espressif.com right now, but please use this link to download the new version of the installer: https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe.

hrecently, 2020-02-25:
Igor, thank you very much! This installer ends with "Setup has finished ESP-IDF Tools on your computer".

closed 2020-02-25 by hrecently