espressif/esp-idf #4824: problem in bluetooth hfp_hf example using HCI

rezakhaji7417, 2020-02-23:
hi.
I'm using [bluetooth hfp_hf example](https://github.com/espressif/esp-idf/tree/v4.0-rc/examples/bluetooth/bluedroid/classic_bt/hfp_hf) , and there is a problem with HCI .
(first, i Apologize, my english is not very good.)

when i config the audio data path to ``HCI`` mode in ``menuconfige`` , I see that  ``"CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_HCI"`` will set "1" or "y" , but in ["bt_app_hf.c"](https://github.com/espressif/esp-idf/blob/v4.0-rc/examples/bluetooth/bluedroid/classic_bt/hfp_hf/main/bt_app_hf.c) file, it is typed ``"CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_HCI"`` and this is wrong.

so, in ["bt_app_hf.c"](https://github.com/espressif/esp-idf/blob/v4.0-rc/examples/bluetooth/bluedroid/classic_bt/hfp_hf/main/bt_app_hf.c) file, I change ``"CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_HCI"`` to ``"CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_HCI"`` , and finally "HCI" work. 
the example code is writed for HCI in "bt_app_hf.c" file, using [ringbuffer](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos_additions.html).

the finall code is here :

```cpp
#if CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_HCI

#define ESP_HFP_RINGBUF_SIZE 3600
static RingbufHandle_t m_rb = NULL;

static void bt_app_hf_client_audio_open(void)
{
    m_rb = xRingbufferCreate(ESP_HFP_RINGBUF_SIZE, RINGBUF_TYPE_BYTEBUF);
}

static void bt_app_hf_client_audio_close(void)
{
    if (!m_rb) {
        return 0;
    }

    vRingbufferDelete(m_rb);
}

static uint32_t bt_app_hf_client_outgoing_cb(uint8_t *p_buf, uint32_t sz)
{
    if (!m_rb) {
        return 0;
    }

    size_t item_size = 0;
    uint8_t *data = xRingbufferReceiveUpTo(m_rb, &item_size, 0, sz);
    if (item_size == sz) {
        memcpy(p_buf, data, item_size);
        vRingbufferReturnItem(m_rb, data);
        return sz;
    } else if (0 < item_size) {
        vRingbufferReturnItem(m_rb, data);
        return 0;
    } else {
        // data not enough, do not read
        return 0;
    }
}

static void bt_app_hf_client_incoming_cb(const uint8_t *buf, uint32_t sz)
{
    if (! m_rb) {
        return;
    }
    BaseType_t done = xRingbufferSend(m_rb, (uint8_t *)buf, sz, 0);
    if (! done) {
        ESP_LOGE(BT_HF_TAG, "rb send fail");
    }

    esp_hf_client_outgoing_data_ready();
}
#endif /* #if CONFIG_BTDM_CONTROLLER_BR_EDR_SCO_DATA_PATH_HCI */

```

in ``bt_app_hf_client_incoming_cb`` function, it receive incoming data from bluetooth and send them into ``m_rb`` buffer.
then, in ``bt_app_hf_client_outgoing_cb`` function, same data in ``m_rb`` buffer will send to bluetooth, so every thing you said, you will be hear that.

Unfortunately, when i programed and run this code to esp32,  the voice i hear has low quality, and the esp32 typed ``"SCO xmit Q overflow, pkt dropped"`` error , and esp32 reset after few seconds.

i think this is a bug in bluetooth api , please help me to fix it. thanks.

AbnerFederer, 2020-02-24:
Hi @rezakhaji7417 

Sorry for not noticing the name change, I will fix it inside as soon as possible.
And for the quality of HF. Do the following maybe well:
1. From my calculation, the value `BTM_SCO_XMIT_QUEUE_THRS` equals `30` will be better. 
2. Change `BTM_SCO_DATA_SIZE_MAX` from `120` to `240`.
3. Comment out the log below is just fine.
```
W (39364) BT_BTM: SCO xmit Q overflow, pkt dropped
W (39364) BT_BTM: stat 10  
```
I have tested the above suggestion on a2dp_sinf_and_hfp_example in ADF.
Please check it out, thank you.

Thanks
Abner

rezakhaji7417, 2020-02-25:
> Hi @AbnerFederer
> 
>
> Thanks for the help, the reset problem was resolved, but unfortunately the sound I still hear is of poor quality. Actually I'm using IDF.
> Please help me and thanks for suggesting another way.
>
> Thanks 
> rezakhaji7417
>

AbnerFederer, 2020-02-26:
Hi, @rezakhaji7417 

We are doing optimization to the host and I think the audio quality is related with the flow control. Please wait for the update of IDF.

If the original issue has been resolved and has a new issue, suggest reporting a new issue. Theoretically, one issue ticket vs one issue. 

Thanks
Abner

rezakhaji7417, 2020-03-02:
Hello @AbnerFederer

Thanks for your help and accountability

This is my last word on this issue. Hope this helps and soon the IDF update will be released, or make another suggestion, because I really need it.

 As the last work in this IDF version, I configured ``i2s`` in the file ``"gpio_pcm_config.c"``, and then put the bluetooth data file in the hci interface in ``"bt_app_hf.c"``. In fact, I write data to ``i2s`` in function ``bt_app_hf_client_incoming_cb`` and also in ``bt_app_hf_client_outgoing_cb`` I read data from i2s. I've put the code below:

```cpp

#if CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_HCI

#define ESP_HFP_RINGBUF_SIZE 3600
//static RingbufHandle_t m_rb = NULL;

static void bt_app_hf_client_audio_open(void)
{
//	m_rb = xRingbufferCreate(ESP_HFP_RINGBUF_SIZE, RINGBUF_TYPE_BYTEBUF);
}

static void bt_app_hf_client_audio_close(void)
{
/*	if (!m_rb) {
        return 0;
    }*/	
//	vRingbufferDelete(m_rb);
}

static uint32_t bt_app_hf_client_outgoing_cb(uint8_t *p_buf, uint32_t sz)
{
	i2s_read(I2S_NUM_0, p_buf, sz, &i2s_bytes_Rd, 0);
	return sz;
    }

static void bt_app_hf_client_incoming_cb(const uint8_t *buf, uint32_t sz)
{
	i2s_write(I2S_NUM_0, buf, sz, &i2s_bytes_Wr, 0);
        esp_hf_client_outgoing_data_ready();
}

#endif /* #if CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_HCI */

```

I have also configured is in the file ``"gpio_pcm_config.c"`` as follows:

```cpp
void app_gpio_pcm_io_cfg(void)
{

#define SAMPLE_RATE     		8000

	i2s_config_t i2s_config_NUM0_Wr_DAC = {
	        .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX,
	        .sample_rate = SAMPLE_RATE,
	        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
	        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,					//2-channels
	        .communication_format = I2S_COMM_FORMAT_I2S,
	        .dma_buf_count = 4,
	        .dma_buf_len = 80,
	        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1						//Interrupt level 1
	    };
	    i2s_pin_config_t pin_config_NUM0_Wr_DAC = {
	        .bck_io_num = 26,
	        .ws_io_num = 25,
	        .data_out_num = 22,
	        .data_in_num = 23												//Not used
	    };
	
	i2s_driver_install(I2S_NUM_0,&i2s_config_NUM0_Wr_DAC,0,NULL);

	i2s_set_pin(I2S_NUM_0,&pin_config_NUM0_Wr_DAC);
	
	i2s_set_clk(I2S_NUM_0,SAMPLE_RATE, I2S_BITS_PER_SAMPLE_16BIT, 2);
	
	vTaskDelay(2000 / portTICK_RATE_MS);
	
	printf("i2s for HCI is installed just now !!!\n");
}

```

The sound quality of the bluetooth received in the ``bt_app_hf_client_incoming_cb`` function is very good, but unfortunately the sound quality of the bluetooth received in ``bt_app_hf_client_outgoing_cb`` is is very poor and almost indistinguishable.

I hope this experience helps and can help you to solve this problem.

Thank you.
rezakhaji7417

AbnerFederer, 2020-03-02:
Hi, @rezakhaji7417 

Thank you very much.
I suspect the poor audio quality is introduced by the mismatching packets pace between host and controller. I'll do my best to fix it as soon as possible.

Thanks
Abner

Alvin1Zhang, 2020-03-02:
@rezakhaji7417 Thanks for reporting and the feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-03-02 by Alvin1Zhang

onyedikilo, 2020-03-09:
@AbnerFederer for the time being can you suggest a fix for the sound quality? Or can you point me to the right direction for me to try and fix the packet pace mismatch? I am using HCI and bt_app_hf_client_incoming_cb quality is great, unfortunately bt_app_hf_client_outgoing_cb is really bad.

Alvin1Zhang, 2020-03-10:
@onyedikilo Thanks for the updates. For the new issue, would you please help file a new GitHub Issue ticket? Thanks.

onyedikilo, 2020-03-10:
@Alvin1Zhang its not a new issue, same problem.

onyedikilo, 2020-03-10:
`if (control_num < 5)
                            {
                                control_num++;
                            }
                            else
                            {
                                control_num = 0;
                                esp_hf_client_outgoing_data_ready();
                            }`

seems to fix the problem for me.

RegginYag, 2020-11-13:
I have a quick question about the "m_rb" ring buffer. How do I actually write data into it, and, in "bt_app_hf.c", why is the                 "esp_hf_client_register_data_callback()" function called before "bt_app_hf_client_audio_open()"?
The function "bt_app_hf_client_audio_open()" creates the ring buffer, so how can "esp_hf_client_register_data_callback()" read and write from the ring buffer before the ring buffer itself is even created?
You can contact me on lukacacictic@gmail.com if you want.  I'd be very grateful for any help. You guys seem to have a lot more experience with ESP32 than me.

jokubasver, 2022-06-05:
@rezakhaji7417 Thanks for the example. It worked for me, but the audio both from the DAC and mic sounded identically robotic. I'm using UDA1334A and INMP441. I've tried different I2S formats (I2S, PCM, MSB), but audio never sounded right. 

I've attached an audio recording: 
[hfp call recording before.zip](https://github.com/espressif/esp-idf/files/8840148/hfp.call.recording.before.zip)

To debug, I opened the audio file in Audacity as 32bit signed PCM - the audio sounds fine, albeit the pitch and speed is wrong. If I do I2S_BITS_PER_SAMPLE_32BIT, then it sounds wrong again. 

The fix for this was to set the I2S DAC to 32bit, and then use i2s_write_expand to expand 16bits to 32bits:
`i2s_write_expand(I2S_NUM_0,(void*) buf, sz, I2S_BITS_PER_SAMPLE_16BIT, I2S_BITS_PER_SAMPLE_32BIT, &i2s_bytes_write, portMAX_DELAY);`

And here's how it sounds now! Nice and clean!
[hfp call recording after.zip](https://github.com/espressif/esp-idf/files/8840150/hfp.call.recording.after.zip)

Now I just need to tackle the microphone, as there is no i2s_read_expand function. Any idea on how to do this?