espressif/esp-idf #4825: http client with port > 3000

chegewara, 2020-02-23:
Hi,
i have simple code for now, basically copy/paste from here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_client.html#application-example

It is working with port 80, but i have issue with connecting to node server with port 3004.
Yes, i am sure the server is working (tested with browser and postman).
Here is log:
```
I (3263) example_connect: IPv4 address: 192.168.43.187
E (21529) HTTP_CLIENT: Connection failed, sock < 0
I (21529) http: HTTP_EVENT_DISCONNECTED
```
ESP-IDF:         ` v4.0-138-g024a2755b`

chegewara, 2020-02-23:
I performed another test. Redirected port 88 to 3004 on server and have the same result.

If i can provide more info or i can do some more tests please let me know.

PS i already changed LWIP max sockets to 16, but this is the only connection opened

shubhamkulkarni97, 2020-02-25:
@chegewara Can you share more details about the server?

Is the server chunked or non-chunked?

chegewara, 2020-02-25:
Its the simplest non-chunked express (nodejs) server.

I just opened port 88 on server, so it is accessible from internet now.
`esp32.eu.org:88/api`
(i am testing it with esp32 on LAN).

shubhamkulkarni97, 2020-02-26:
@chegewara I tried connecting to python, nodes, openssl servers with `port > 3000` and it worked for me.
Here is the log:
```
I (5137) example_connect: IPv4 address: 10.0.1.166
I (5147) example_connect: IPv6 address: fe80:0000:0000:0000:b6e6:2dff:feeb:171d
I (5147) HTTP_CLIENT: Connected to AP, begin http example
I (5207) HTTP_CLIENT: HTTP GET Status = 200, content_length = 21
I (5207) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```

Can you run `wget http://<host_ip>:<server_port>` successfully?

chegewara, 2020-02-26:
Thanks, but like i said later, initially i thought its issue with port, but then i realized its something wrong with connecting to nodejs(express) server. I will try to dig more, maybe its problem because i have this server serving on 2 ports.
HTTPS for external access with valid certificate 
and 
HTTP for LAN access for development purpose.

Anyway thanks for help.

shubhamkulkarni97, 2020-03-02:
@chegewara Can we close this issue?

chegewara, 2020-03-03:
From my point of view the issue is still not resolved. Maybe it works in most cases, but my server configuration may be specific and eventually bug in esp-idf does not allow to connect.

Could you or someone else to confirm it is possible to connect with my server?
`http://esp32.eu.org:88/api`

Like i said, i can connect from browser, postman and with curl and wget from terminal, so its most likely not a problem with my server.

shubhamkulkarni97, 2020-03-03:
> Could you or someone else to confirm it is possible to connect with my server?
> `http://esp32.eu.org:88/api`

Yes, I can connect to this server. These are the logs:

```
Connecting to http://esp32.eu.org:88/api
I (4120) HTTP_CLIENT: HTTP GET Status = 200, content_length = 23
I (4120) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```

chegewara, 2020-03-03:
Thanks. Then its probably something with my code.
Just to confirm, i am using branch v4.0

chegewara, 2020-03-04:
Today i updated esp-idf and now its working with release v4.0 and v4.1.

closed 2020-03-04 by chegewara