espressif/esp-idf #4827: pthread_cond_timedwait() - Timeout overflows for larger wait times

alemariusnexus, 2020-02-24:
## Environment

IDF version: Latest master (7092613eea299626a103925c0c1ff5587297d6c6 at the time of writing).

Rest should be irrelevant.

## Problem Description

pthread_cond_timedwait() does not handle "large" timeouts (> ~25 days) correctly.

The problem starts with the following line in [components/pthread/pthread_cond_var.c](https://github.com/espressif/esp-idf/blob/7092613eea299626a103925c0c1ff5587297d6c6/components/pthread/pthread_cond_var.c#L118):

`timeout_msec = (diff_time.tv_sec * 1000) + (diff_time.tv_usec / 1000);`

Since tv_sec is of type time_t (which seems to be int32_t internally), multiplying it by 1000 will lead to an overflow for any timeouts larger than ~25 days (INT32_MAX/1000 seconds). This could be fixed by casting either factor to a larger integer type, but the problem continues because both timeout_msec and timeout_ticks (used later) are 32 bit types, so they can overflow as well.

While it might not seem useful to have such large timeouts, some of my code uses INT32_MAX for tv_sec to say that it wants to wait "indefinitely" (I don't use pthread_cond_wait() because sometimes I _do_ want to timeout earlier). In the IDF implementation, this would lead to an overflow to a negative timeout, making pthread_cond_timedwait() wake up immediately. To make it work, I would need to use another value specifically to work around this problem, making my code less portable.

I think the most sensible fix would be to check if any of the calculations would overflow, and instead use the maximum permissible value in those cases.

X-Ryl669, 2020-02-25:
Isn't 25 days already almost "indefinitely" for a waiting a condition ?
Can you define your `Indefinitely` value as `INT32_MAX / 1000` ?

alemariusnexus, 2020-02-25:
That's what I'm currently doing as a workaround, but it's not an elegant solution. It's still a hassle if you want to port existing code, and I'd still consider it a bug if the function wakes up immediately if I provide anything close to INT32_MAX.

X-Ryl669, 2020-02-26:
Please notice that overflowing code is often platform specific in C/C++ (understand: not portable or undefined). They might be platforms where `INT32_MAX * X = INT32_MAX` (yet, I never heard of any). 
The fact that you are using a signed integer to store what semantically should be a unsigned value, is probably the issue here. If you want X*Y to keep the original sign, use unsigned arithmetic. In that case, `UINT32_MAX * 1000` is `0xfffffc18` which is almost like `UINT32_MAX`.
Anyway, it's very unlikely the underlying code will change to fit your requirement/convention.

uLipe, 2020-02-27:
Hello @alemariusnexus, as @X-Ryl669 already pointed, the overflow is typically platform dependent or undefined.

I see you are expecting the same effect of portMAX_DELAY constant when passed to any FreeRTOS blocking function, it will wait for operation indefinitely. 

In the case of pthread_cond_wait_timed(), it expects the caller already has a reasonable time to wait (where ~25 days of wait falls out of reasonable). Otherwise, it is currently possible to trigger the blocking indefinitely behavior by simply making the timespec *to parameter equals to NULL.

uLipe, 2020-03-30:
Hi Folks, seems this thread got inactive, do you have any more concerns of the current `pthread_cond_timedwait()`? Otherwise, this thread will be closed.

Thanks :)

Alvin1Zhang, 2020-04-02:
@alemariusnexus Thanks for reporting and feedback, will close due to short of feedback, feel free to reopen with details. Thanks.

closed 2020-04-02 by Alvin1Zhang