espressif/esp-idf #4828: "file_serving" example in esp-idf examples not working

GMasciaRDT, 2020-02-24:
Hello,

I use the ESP32 with the ESP-IDF SDK (release/v4.0) in custom device that among the other things relies on the ESP32 to create  a WiFi access point and a File Server to:

- Upload OTA FW updates 
- Download data files containing GPS and IMU data (several MB each) from an SD card

I'd like t use the "file_serving" example in "esp-idf/examples/protocols/http/server" but I encountered several issues:
- Very often the transfer fails
- When I'm able to successfully transfer the data, the upload/download speed is ridiculously low (~10KBps)
- Sometimes I get timeout errors even though the ESP32 is not actually doing anything

I attach a picture showing a typical error when uploading a FW binary file.
[image]

I'm stuck with this since a long time and I've not been able to find any fix to the issue on the internet. 
I've already contacted the Espressif team and they confirmed me they get the same errors when running the example. I've been told that they're working on the issue but this could take some time.
 
Do anybody else experienced the same problem? Could you please help me with that?

shubhamkulkarni97, 2020-02-28:
@GMasciaRDT 

Total partition size is 896321 bytes, of which 768060 bytes are used. You are trying to upload a file of size 148k, which exceeds the total partition size. That is the reason you are getting the error.

Increasing partition size of spiffs storage in `partitions_example.csv` would solve your issue.

GMasciaRDT, 2020-02-28:
Thank you @shubhamkulkarni97 !

I figured that out and in fact I used the command esp_spiffs_format to format the spiffs. 
Then I used the SD card instead of spiffs, because in the project I am working on, the server reads/writes files from/to a SD card.

However, I'm still trying to improve the upload and download speed. Increasing SCRATCH_BUFSIZE to 65536 I reached 200kBps for upload and 440kBps for download. It could be ok for upload, but I would like to keep improving the download speed, since the files can be tens of MB each. 
Do you have any suggestions for this matter?

shubhamkulkarni97, 2020-03-02:
@GMasciaRDT Can you try the following changes in sdkconfig.
```
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_RX_BA_WIN=32
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
```

Using these changes will increase the throughput but will increase memory consumption.

GMasciaRDT, 2020-03-04:
Thank you for your help @shubhamkulkarni97 .

I just made the changes in the sdkconfig file and tested.
The first time I just executed "make flash", and the throughput was almost the same. 
Then I executed "make clean" followed by "make flash" and I got the following error:
httpd_txrx: httpd_sock_err: error in recv : 11

In general, file downloads fails, whereas file upload complete with a throughput much worse than before.

shubhamkulkarni97, 2020-05-26:
@GMasciaRDT Sorry for late reply, I somehow missed out this issue.

Is your issue resolved?

If not, can you try modifying the sdkconfig in the ESP-IDF file_serving example and check if it works?

If this doesn't solve your issue, can you share sample code and sdkconfig to reproduce this issue?

shubhamkulkarni97, 2020-07-10:
@GMasciaRDT Any updates on the issue?

Alvin1Zhang, 2020-07-20:
@GMasciaRDT Thanks for reporting, will close due to short of updates, feel free to reopen with more details/updates. Thanks.

closed 2020-07-20 by Alvin1Zhang

Rabiraj-0204, 2021-10-20:
I am trying to download a file , using Simple HTTP File Server Example. But when I select upload option in the http server. Then the debug log , I am able to get the message as : 
I (1279073) file_server: Remaining size : 24032
I (1279133) file_server: Remaining size : 19712
I (1279193) file_server: Remaining size : 15392
I (1279263) file_server: Remaining size : 11072
I (1279383) file_server: Remaining size : 5328
I (1279473) file_server: File reception complete

But According to the manual , which is in " https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/file_serving/README.md "

I am trying to download the file , which I downloaded through the HTTP server , as in above example.

I am getting a error as : 
E (2987433) file_server: Failed to stat file : /spiffs/C:/Users/myfile_copy.html
W (2987433) httpd_txrx: httpd_resp_send_err: 404 Not Found - File does not exist
W (2987443) httpd_uri: httpd_uri: uri handler execution failed

Woud anyone help with this.

I am trying to perform OTA in AP mode . For that I have combined Wifi AP example , Simple HTTP File Server Example  , OTA example.
As of now , Wifi is in Stationary mode. So Once I get the get the file , using esp_ota_begin function I will send that file.