espressif/esp-idf #4829: spi_flash wrong detected size when upgrading to v4.0

galah92, 2020-02-24:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.0
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm getting `spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed` while I got 8MB and used specifically the following to flash my app:

    python -m esptool -p COM5 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 8MB --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x15000 build\ota_data_initial.bin 0x20000 build\MyProject.bin

Note this happens in the process of updating from v3.3.1 to v4.0.

## Debug Logs

```
    rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0018,len:4
    load:0x3fff001c,len:6840
    load:0x40078000,len:14076
    load:0x40080400,len:4304
    entry 0x400806e8
    I (71) boot: Chip Revision: 1
    I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
    I (39) boot: ESP-IDF v4.0-dirty 2nd stage bootloader
    I (39) boot: compile time 14:55:41
    I (40) boot: Enabling RNG early entropy source...
    I (45) boot: SPI Speed      : 40MHz
    I (49) boot: SPI Mode       : DIO
    I (53) boot: SPI Flash Size : 8MB
    I (57) boot: Partition Table:
    .....................................
    I (120) boot: End of partition table
    I (124) boot_comm: chip revision: 1, min. application chip revision: 0
    I (132) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0xfcaa0 (1034912) map
    I (511) esp_image: segment 1: paddr=0x0011cac8 vaddr=0x3ffbdb60 size=0x03438 ( 13368) load
    I (516) esp_image: segment 2: paddr=0x0011ff08 vaddr=0x40080000 size=0x00108 (   264) load
    0x40080000: _WindowOverflow4 at C:/esp-idf/components/freertos/xtensa_vectors.S:1778

    I (518) esp_image: segment 3: paddr=0x00120018 vaddr=0x400d0018 size=0xdeb54 (912212) map
    0x400d0018: _stext at ??:?

    I (853) esp_image: segment 4: paddr=0x001feb74 vaddr=0x40080108 size=0x002f8 (   760) load
    0x40080108: _WindowOverflow12 at C:/esp-idf/components/freertos/xtensa_vectors.S:1927

    I (853) esp_image: segment 5: paddr=0x001fee74 vaddr=0x40080400 size=0x1a768 (108392) load
    I (921) boot: Loaded app from partition at offset 0x20000
    I (921) boot: Disabling RNG early entropy source...
    I (922) cpu_start: Pro cpu up.
    I (925) cpu_start: Application information:
    I (930) cpu_start: Project name:     MyProject
    I (935) cpu_start: App version:      410:0.0.10
    I (940) cpu_start: Compile time:     Feb 24 2020 14:55:23
    I (946) cpu_start: ELF file SHA256:  fb90f8bdc123454f...
    I (952) cpu_start: ESP-IDF:          v4.0-dirty
    I (958) cpu_start: Starting app cpu, entry point is 0x40081200
    0x40081200: call_start_cpu1 at C:/esp-idf/components/esp32/cpu_start.c:272

    I (945) cpu_start: App cpu up.
    I (968) heap_init: Initializing. RAM available for dynamic allocation:
    I (975) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
    I (981) heap_init: At 3FFD3488 len 0000CB78 (50 KiB): DRAM
    I (987) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
    I (994) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
    I (1000) heap_init: At 4009AB68 len 00005498 (21 KiB): IRAM
    I (1006) cpu_start: Pro cpu start user code
    I (1025) spi_flash: detected chip: generic
    I (1026) spi_flash: flash io: dio
    E (1026) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.
    abort() was called at PC 0x40081142 on core 0
```

Note I get `I (53) boot: SPI Flash Size : 8MB` at the beginning and `Detected size(4096k)` at the end.

ginkgm, 2020-02-24:
Hi @galah92 ,

The default flash size of WROOM32-D is 4MB, which can be founded here: https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf.

In this case, then the esp_flash API detects the correct ID and size. Which means the flash size configured in the binary header (8MB) is incorrect. You may need to fix it in the menuconfig: https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esptoolpy-flashsize.

But if this is not the case, you get a [customized WROOM32-D](https://www.espressif.com/sites/default/files/documentation/espressif_products_ordering_information_en.pdf) with 8MB flash, then it's wired. Could you please run `esptool.py flash_id` to see the flash ID and post it here?

galah92, 2020-02-24:
> Hi @galah92 ,
> 
> The default flash size of WROOM32-D is 4MB, which can be found here: https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf.
> 
> In this case, then the esp_flash API detects the correct ID and size. Which means the flash size configured in the binary header (8MB) is incorrect. You may need to fix it in the menuconfig: https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esptoolpy-flashsize.
> 
> But if this is not the case, you get a [customized WROOM32-D](https://www.espressif.com/sites/default/files/documentation/espressif_products_ordering_information_en.pdf) with 8MB flash, then it's wired. Could you please run `esptool.py flash_id` to see the flash ID and post it here?

    esptool.py v2.7
    Found 1 serial ports
    Serial port COM5
    Connecting....
    Detecting chip type... ESP32
    Chip is ESP32D0WDQ5 (revision 1)
    Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
    Crystal is 40MHz
    MAC: bc:dd:c2:c4:3e:84
    Uploading stub...
    Running stub...
    Stub running...
    Manufacturer: 20
    Device: 4016
    Detected flash size: 4MB
    Hard resetting via RTS pin...

You're correct - I got 4MB of flash. Thank you for the help.

Alvin1Zhang, 2020-02-27:
@galah92 Glad that your issue is resolved. Feel free to reopen if the issue still happens or file another ticket for new issue. Thanks.

closed 2020-02-27 by Alvin1Zhang