espressif/esp-idf #4830: Can't read any byte from serial

romocitto88, 2020-02-24:
## Environment

- Development Kit:      ESP32-Dev-Kit
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.0
- Build System: idf.py
- Operating System: Windows
- Using an IDE?: Yes PlatformIO
- Power Supply: USB

## Problem Description

I tried all possible example code but I can't get any byte read from uart (I'm using uart2). 
Trasmitting is ok.
Below the code that i'm using.

### Code to reproduce this issue

```cpp

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/uart.h"
#include "driver/gpio.h"

unsigned char dati_condivisi[100];
QueueHandle_t drvuart_queueUart2;

#define DRVUART_UART2_BAUDRATE 115200
#define DRVUART_UART2_DATABITS UART_DATA_8_BITS
#define DRVUART_UART2_PARITY UART_PARITY_DISABLE
#define DRVUART_UART2_STOPBITS UART_STOP_BITS_1
#define DRVUART_UART2_FLOW_CTRL UART_HW_FLOWCTRL_DISABLE
#define DRVUART_UART2_RX_FLOW_CTRL_THRESH 122

#define DRVUART_UART2_TX_PIN GPIO_NUM_17
#define DRVUART_UART2_RX_PIN GPIO_NUM_16
#define DRVUART_UART2_RTS_PIN UART_PIN_NO_CHANGE
#define DRVUART_UART2_CTS_PIN UART_PIN_NO_CHANGE

#define DRVUART_UART2_TX_BUFFER_SIZE 1024
#define DRVUART_UART2_RX_BUFFER_SIZE 1024
#define DRVUART_UART2_QUEUE_DEPTH 10
#define DRVUART_UART2_QUEUE_POINTER &drvuart_queueUart2
#define DRVUART_UART2_ALLOCATE_FOR_INTRRUPT 0

void uart2_task(void *pvParameters)
{
    while(1)
    {
        uint16_t length;
        uart_get_buffered_data_len(UART_NUM_2,&length);
        if(length)
        {
            uart_read_bytes(UART_NUM_2,dati_condivisi, 10, 100);
            printf("Data ok\n");
        }
    }
    vTaskDelete(NULL);
}

void app_main() 
{
    uart_config_t uart_config = {
          .baud_rate = DRVUART_UART2_BAUDRATE,
          .data_bits = DRVUART_UART2_DATABITS,
          .parity    = DRVUART_UART2_PARITY,
          .stop_bits = DRVUART_UART2_STOPBITS,
          .flow_ctrl = DRVUART_UART2_FLOW_CTRL
    };
    //uart_config.rx_flow_ctrl_thresh=DRVUART_UART2_RX_FLOW_CTRL_THRESH;
    uart_param_config(UART_NUM_2, &uart_config);
    uart_set_pin(UART_NUM_2, DRVUART_UART2_TX_PIN, DRVUART_UART2_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_driver_install(UART_NUM_2, DRVUART_UART2_RX_BUFFER_SIZE, DRVUART_UART2_TX_BUFFER_SIZE, DRVUART_UART2_QUEUE_DEPTH, DRVUART_UART2_QUEUE_POINTER, DRVUART_UART2_ALLOCATE_FOR_INTRRUPT);
    xTaskCreate(uart2_task, "uTask", 2048, (void*)UART_NUM_2, 8, NULL);
}

```

koobest, 2020-02-25:
Hi,
Please call `uart_param_config` before calling `uart_driver_install`.
```
void init(void) {
    const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        //.source_clk = UART_SCLK_APB,
    };
    // We won't use a buffer for sending data.
    uart_param_config(UART_NUM_1, &uart_config);
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 2, 0, 0, NULL, 0); 
    uart_set_pin(UART_NUM_1, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}
```

romocitto88, 2020-02-25:
Thank you problem is solved.

closed 2020-02-25 by romocitto88