espressif/esp-idf #4833: ESP-IDF crashes while sending more than 128bytes via UART1 using uart_write_bytes() function

sumit809, 2020-02-25:
## Environment

- Development Kit:     [ESP32-PICO-Kit]
- Kit version (for PicoKit): [v4]
- Module or chip used:  [ESP32-PICO-D4]
- IDF version : 3.30300.190916 (3.3.0)
- Build System:         [idf.py]
- Compiler version: 2.50200.80 (5.2.0)
- Operating System:     [macOS]
- Using an IDE?: [Yes (PlatformIO)]
- Power Supply:         [USB]

## Problem Description

I'm trying to send number of APs available, their MAC addresses and their respective RSSI in one string all at once.
The code works fine if I send the string by breaking it into 128 Bytes, but crashes when I try to send the entire thing

### Expected Behavior
The code works fine if I send the string by breaking it into 128 Bytes

### Actual Behavior
crashes when I try to send the entire thing

### Code to reproduce this issue

<script src="https://gist.github.com/sumit809/603d7209a13614dab6b02df1d49f8a12.js"></script>

## Debug Logs

```
<script src="https://gist.github.com/sumit809/368823fc8b0bf379d59e928756953794.js"></script>

```

## Other items
it contains main.c and sdkconfig file
[git uart issue Codes.zip](https://github.com/espressif/esp-idf/files/4248865/git.uart.issue.Codes.zip)

koobest, 2020-02-25:
Hi,  @sumit809
Can you provide a log file? I cannot open the link.
thanks !!

sumit809, 2020-02-25:
[log.txt](https://github.com/espressif/esp-idf/files/4249192/log.txt)
please have a look at it...

koobest, 2020-02-25:
Hi, @sumit809
I get stack overflow error,  please use  `malloc` to allocate memory for a  `s2` and `string`.
```
        const size_t N = sizeof(macArray) / sizeof(*macArray);
        int len1 = len * 3;
        // char s1[] = "ESP:SCANW:";
        char *s2 = malloc(len1);
        int n;
        for (size_t i = 0, j = 0;
             i < N && (n = snprintf(s2 + j, sizeof(s2) - j, "%d", macArray[i])) > 0;
             i++)
        {
            j += n;
            len1 = j;
        }
        int len2 = len1 + 11;
        char* string = malloc(len2);
        ...
        free(s2);
        free(string);
```

sumit809, 2020-02-25:
Hi, @koobest 
This worked and prevented from crashing. Thanks for this!

However, I'm facing new problem i.e., the string variable is not able to hold more than 13 bytes at any case. I tried to send data by UART for two times... and these are the results

### UART0 output:
Number of access points found: 11
total bytes in string variable = 13
total loops=1
remaining bytes=13

### UART1 output:
ESP:SCANW:111

----------------------------

### UART0 output:
Number of access points found: 12
total bytes in string variable = 13
total loops=1
remaining bytes=13

### UART1 output:
ESP:SCANW:121

Can you tell me why this is happening. Also, why did allocating memory using malloc prevented from crashing?

sumit809, 2020-02-26:
Hi, @koobest 
Actually the problem is I can't send more than 128 bytes at once using uart_write_bytes() 
You can reproduce this issue by simply creating a string consisting of more than 128 bytes and try to send it using uart_write_bytes via UART1.

Also, I found out the reason that why it had stopped crashing after using malloc()... the reason was it was not able to cross the 128 bytes length.

Please guide if you have any Idea... Or anybody who is reading this thread.
Thanks in Advance...!

costaud, 2020-02-26:
@sumit809  Are you sure you want to use your own ISR handler ?  Strongly suggest not do so. 
If you really want to use your own ISR handler, strongly suggest not to mix up your ISR and the default driver API, like uart_write_bytes. Since such APIs are based on the internal ISR handler. 

Back to this issue, you can see this in the log, 
> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

which indicates that the code get stuck in your ISR handler. Probably because your code won't handler all the UART interrupt which are enabled by the default driver.(Why 128 bytes? Because that could probably triggered a fifo overflow interrupt)

My suggestion is that please read the uart example in IDF first and use the default uart driver in IDF. If the default uart driver can't meet your requirements, please let us know, we would try to provide solution. If you want to write your own driver, please don't call the default driver APIs.

sumit809, 2020-02-26:
Hi @costaud 
Actually I want the ISR for receiving part only and not for the transmitting part. I had written the ISR assuming that it has nothing to do with FIFO overflow interrupt. So, what can I do in this case?

sumit809, 2020-02-27:
For now I have avoided using my own ISR and that solves this issue... But I would still want to know how are those internal ISRs are written. If anybody can show the examples... that will be great.
Thanks

closed 2020-02-27 by sumit809