espressif/esp-idf #4835: Flash encryption

varunsavita, 2020-02-25:
I selected 
 Enable flash encryption on boot (READ DOCS FIRST)   under security feature.
 Bootloader log verbosity (Info) under bootloader config.
 Partition Table (Single factory app, no OTA)
  (0x9000) Offset of partition table   
** Error **
**idf.py encrypted-app-flash monitor** 

Serial port COM3
Connecting........_____....._____....._____....._____.
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:e9:4b:09
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB

WARNING: - compress and encrypt options are mutually exclusive
Will flash uncompressed
Writing at 0x00024000... (20 %)
A fatal error occurred: Timed out waiting for packet header
esptool.py failed with exit code 2

varunsavita, 2020-02-25:
[image]

Could you please to help me to understand the question mark on pic

negativekelvin, 2020-02-26:
Question mark means value is read protected.  Encryption is on.

varunsavita, 2020-02-26:
@negativekelvin 
** Error **
idf.py encrypted-app-flash monitor

Serial port COM3
Connecting........................
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:e9:4b:09
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB

WARNING: - compress and encrypt options are mutually exclusive
Will flash uncompressed
Writing at 0x00024000... (20 %)
A fatal error occurred: Timed out waiting for packet header
esptool.py failed with exit code 2

negativekelvin, 2020-02-26:
encrypted-app-flash doesn't work if DISABLE_DL_ENCRYPT=1

projectgus, 2020-02-26:
That's right, this target depends on configuring flash encrption in "Development Mode" in the project. Was that the case here?

We need to add support in esptool.py to check the flash encryption efuse status before the encrypted upload begins.

varunsavita, 2020-03-19:
I used flash encryption then is it possible?
Someone can read the sketch code from my esp32 module?

projectgus, 2020-04-01:
Hi @varunsavita ,

If flash encryption is enabled in Release mode (which seems to be the case in the screenshot you posted of espefuse output), then it's not possible for someone to easily read out the flash contents including the firmware binary.

Does this answer your question?

Angus

varunsavita, 2020-04-07:
Thank you for your support.
Yes, I got

projectgus, 2020-04-07:
Hi @varunsavita ,

Good to hear. I'm going to close this issue, then.

For future reference, these kind of questions about ESP-IDF (that aren't reporting an issue or a missing feature) are best asked over at https://esp32.com forum .

Angus

closed 2020-04-07 by projectgus

varunsavita, 2020-04-07:
The first time, I upload encrypted flash on then I saw as an image 
Now I want to upload code.
So how to do?
printing the error flash read err, 1000 - meant that flash encrypted enable
how to change FLASH_CRYPT_CNT  it?

projectgus, 2020-05-04:
@varunsavita Sorry I didn't see this earlier. Did you end enable Flash Encryption in Release mode or Debug mode?