espressif/esp-idf #4836: Stop boot rom logs , panic handler logs on UART 0 on Esp-Wroom-32

manojsai17, 2020-02-25:
We interfaced simcom gsm module on uart0 . 
Please suggest a way to stop boot rom logs , panic handler logs on UART 0 on Esp-Wroom-32.
As per few posts we have no option to pull down on GPIO 15 .
Please suggest a way in a software way to stop logs on Uart 0 .

igrr, 2020-02-25:
Hi @manojsai17, if you can't pull down GPIO15, then unfortunately the ROM logs which follow a power-on reset can not be stopped.

You can disable the rest of the output by setting the following options:
CONFIG_ESP_CONSOLE_UART_NONE=y
CONFIG_ESP32_PANIC_SILENT_REBOOT=y

However this arrangement will make your development flow quite hard, as you will not be getting feedback when something goes wrong in the program. I would suggest connecting the GSM module to some other GPIOs. UART peripheral can be connected to other GPIOs using the GPIO matrix, so if you have any pair of GPIOs available, you can configure UART1 to use them.

Also as stated in the issue template which was presented to you when you opened this issue, please use esp32.com forum for the "how do I..." kind of questions.

chegewara, 2020-02-25:
Using bootloader as component and editing it to not send any logs is not an option?

igrr, 2020-02-25:
@chegewara The output of the 2nd stage bootloader (which is part of IDF) can be disabled using the CONFIG_ESP_CONSOLE_UART_NONE=y option. There is no need to edit the source code.
However the 1st stage bootloader is located in the mask ROM (i.e. it can not be modified), and its output can only be affected by GPIO15 state at reset.

Alvin1Zhang, 2020-02-26:
@igrr Thanks for the notes. @manojsai17 Thanks for reporting, as for "how do I..." kind of questions, please help use esp32.com forum, will close this ticket for now, thanks.

closed 2020-02-26 by Alvin1Zhang