espressif/esp-idf #4837: Panic while calling time() in 2 tasks simultaneously

X-Ryl669, 2020-02-25:
## Environment

- Development Kit:      [ESP32-Wrover]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v4.1-dev-153-g10dec4e
- Build System:         [Make]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

I'm experiencing an issue with a recent esp-idf. 

The issue I have happens with 2 tasks both calling `time()` at the same time (pun intended). 
Since time internally calls `gettimeofday_r()` as I understand it, it should work, but it panics instead with this stacktrace:
```
Current time: 2020/02/25 16:47:56 (for 1582649270)
abort() was called at PC 0x40153d95 on core 0
0x40153d95: pthread_include_pthread_local_storage_impl at ~/esp/esp-idf/components/pthread/pthread_local_storage.c:262

ELF file SHA256: 9d6942b05dc1a62601a184aa5aabb0a7ecfb06703d5d38486af8970f00ac97bf

Backtrace: 0x4008c7bf:0x3ffb2620 0x4008cbd8:0x3ffb2650 0x40153d95:0x3ffb2680 0x40161228:0x3ffb26b0 0x4010d98f:0x3ffb26d0 0x400d0a83:0x3ffb2760
0x4008c7bf: invoke_abort at ~/esp/esp-idf/components/esp32/panic.c:161

0x4008cbd8: abort at ~/esp/esp-idf/components/esp32/panic.c:176

0x40153d95: pthread_include_pthread_local_storage_impl at ~/esp/esp-idf/components/pthread/pthread_local_storage.c:262

0x40161228: copysignf at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libm/common/sf_copysign.c:34

0x4010d98f: app_main at ~/esp/nvse_app/main/MeasureDB.hpp:83
 (inlined by) app_main at ~/esp/nvse_app/main/main.cpp:164

0x400d0a83: main_task at ~/esp/esp-idf/components/esp32/cpu_start.c:576
```

The code in MeasureDB.hpp:83 reads:
```
{ 
time_t n; 
::time(&n); // HERE
[...] 
```
Just before the panic, I had these logs:
```
Current time: 2020/02/25 16:47:56 (for 1582649270)
```
Generated by the main task:
```
    void getUTCTime(struct tm * timeInfo) const
    {
        time_t now;
        time(&now);
        gmtime_r(&now, timeInfo);
    }
    void printCurrentTime(const char * prefix) const
    {
        struct tm ti;
        getUTCTime(&ti);
        time_t now; time(&now);
        printf("%s%d/%02d/%02d %02d:%02d:%02d (for %u)", prefix, ti.tm_year + 1900, ti.tm_mon+1, ti.tm_mday, ti.tm_hour, ti.tm_min, ti.tm_sec, now);
    }
```
The function `printCurrentTime` is called whenever the SNTP client callback fire (that was installed via `::sntp_set_time_sync_notification_cb` function)
 
So to sum this up:

1.  SNTP task is triggerred and update the time 
2. It calls the callback registered with`sntp_set_time_sync_notification_cb`
3. The callback notify the main task
4. The main task calls `time` and `gmtime_r`

Simultaneously, another task is doing:

1. Calling `time`
2. Core panics with unintelligible backtrace.

### Expected Behavior
There shouldn't be any panic

### Actual Behavior
It panics.

### Steps to reproduce
Please read above.

Alvin1Zhang, 2020-02-26:
@X-Ryl669 Thanks for the detailed report. We will look into. Thanks.

KonstantinKondrashov, 2020-05-13:
Hi @X-Ryl669!
Yes, there is gettimeofday() inside time(). The log does not show any issue with time(). I am not sure that it is linked to it. We have a similar unit test to cover the  `2 tasks simultaneously use gettimeofday()` case and I also tried to reproduce with your part of code but I can not reach your result.
Could you share a code that reproduces it?
Thanks.

X-Ryl669, 2020-05-13:
Well, I'm not sure I can revert to that far in time. 
It happened very sparsely (the timing for this bug is very hard to reproduce, since both task must call `time()` at the same time). 

Looking at the code it's clear there is a logic flaw with time adjustment in the file `newlib/time.c`:
If you look into `adjtime` method, it's taking the lock `s_adjust_time_lock` (line 207), then call `adjust_boot_time()` which, in turns, locks `s_boot_time_lock` while calling `get_boot_time()`.

Thus, if any task is calling `adjtime()` function, the lock order is `s_adjust_time_lock` then `s_boot_time_lock`. If any other task is calling `get_boot_time()` or `gettimeofday()` then it'll lock `s_boot_time_lock` thus blocking `adjtime` **midway** while it's doing its work and time adjustment will be wrong.

Also, if I follow the complex code of newlib, calling time() [in fact calls `gettimeofday_r`](https://github.com/eblot/newlib/blob/master/newlib/libc/time/time.c), and `gettimeofday_r` is using the `_REENT` macro which, in turns, allocate a `_reent` structure on the FreeRTOS task's TCB. If the tasks has died as shown [here](https://github.com/espressif/esp-idf/blob/ad60568151ae3226375ddb05f33c067213048404/components/freertos/tasks.c#L2120), it returns a pointer to a global structure.

I wonder if both task (LWIP's SNTP task and main task) somehow managed to use the same  `_reent` structure and thus corrupted their state (for example, one task using the `_reent` struct of the other task, while the other task is being deleted)
It's very hard to debug since `time(time_t)` and newlib is hidden in ROM and not debuggable.

In the end, I've changed my code not to call any time related function in the callback and simply post an event to the main task's queue.

KonstantinKondrashov, 2020-05-13:
Thank you for sharing your vision.
I agree with your concern about it is slightly confused but if you see at [the diagram of call](https://www.yworks.com/yed-live/?file=https://gist.githubusercontent.com/KonstantinKondrashov/019f97bb9d64aca3dbad0fca517e96cd/raw/f291c596b08b03042484d6e37c452b02bc988574/bpmn6) it does not show interlocks. It has the order of a lock s_adjust_time_lock -> s_boot_time_lock  or a single lock but does not have another order.

> If any other task is calling get_boot_time() or gettimeofday() then it'll lock s_boot_time_lock thus blocking adjtime midway while it's doing its work and time adjustment will be wrong.

The adjtime() method adds a delta time to the boot time (the boot time is not changed often and it is not important when we start to do it, right?!). We know how many usec (adjtime_total_correction) we need to add at all. The accuracy should not be loose in this case.

I will try to look at it more closely to find something.

Did your code use `adjtime ()`?

projectgus, 2020-05-19:
Hi @X-Ryl669 ,

I have a couple of other observations and some questions:

The "unintellegible" backtrace in the panic handler probably indicates stack corruption - addresses 0x40161228 and 0x40153d95 are part of the app binary in flash, not ROM, but there's no possible call sequence between them. It's likely that something which ran recently before the crash has overflowed a buffer on the stack.

Setting the [stack checking mode to OVERALL](https://docs.espressif.com/projects/esp-idf/en/release-v4.1/api-reference/kconfig.html#config-compiler-stack-check-mode) and [setting the stack canary mode to "canary bytes"](https://docs.espressif.com/projects/esp-idf/en/release-v4.1/api-reference/kconfig.html#config-freertos-check-stackoverflow) may help get a crash closer to the point in time where any stack corruption occurs. (Note these two are for different types of problems: stack checking catches the case where a buffer on the stack overflows due to a bug in the code itself, canary bytes check the case where the code is correct but the stack size allocated for the task which runs the code is insufficient.)

Bear in mind the SNTP callback runs in the context of the LWIP task, and the default stack size for this task is only 3KB. If doing "heavy lifting" in the callback (even the `printf()` call with the long format string may be too much), then suggest either [increasing this value](https://docs.espressif.com/projects/esp-idf/en/release-v4.1/api-reference/kconfig.html#config-lwip-tcpip-task-stack-size) or - better - move work out of the callback so the callback triggers processing in the the main task.

If you still have reason to believe this is a bug in ESP-IDF, could you please share the full SNTP callback that calls through to `printCurrentTime()` and then crashes? Or ideally a complete minimal program which exhibits the crash?

Thanks,

Angus

X-Ryl669, 2020-05-19:
Hi Angus,

  Thanks for you advice. As I said earlier, since Feb, our code has evolved a lot, and it'll be hard to come back to that time. The solution, by the time, was to simply post an event in the callback to the main task and have the main task react to the event so time-related stuff all happened in a single task.

What you've described is probably right, the corruption displayed above does make sense with what we had observed. We have stack canary already enabled. I'm not 100% sure, but it seems we already had stack checking mode set to overall in Compiler Options.

I'm going to close this issue as I'm not able (willing) to reproduce it anymore. Let's hope it serves someone with a similar behavior and the explanations above will helps them too.

Thanks!

closed 2020-05-19 by X-Ryl669

projectgus, 2020-05-19:
@X-Ryl669 Thanks for the update and glad your existing solution hasn't given you any trouble (posting an event is this way is probably the best model in any case). Feel free to open a new issue if you run across something like this again.