espressif/esp-idf #4838: Set uart baud_rate to 921600 does not work with UART_SCLK_REF_TICK

AxelLin, 2020-02-26:
Module or chip used: ESP32-WROOM-32
IDF version: v4.2-dev-432-gc65b67115a52
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

Test using examples/peripherals/uart/uart_echo and change baud_rate and source_clk settings as below:
.baud_rate = 921600,
.source_clk = UART_SCLK_REF_TICK,

Build and run the application, the output shows garbage message.
In this test, CONFIG_PM_ENABLE is not set.

Below tests for reference:
Test Fail: .baud_rate = 921600 && .source_clk = UART_SCLK_REF_TICK (show garbage messages)
Test OK: .baud_rate = 115200 && .source_clk = UART_SCLK_REF_TICK
Test OK: .baud_rate = 921600 && .source_clk = UART_SCLK_APB
TEST OK: .baud_rate = 115200 && .source_clk = UART_SCLK_APB

PS. My main intention was to test CONFIG_PM_ENABLE with different source_clk and baud_rate.
But I hit this issue even when CONFIG_PM_ENABLE is disabled.

koobest, 2020-02-26:
Hi,
When using ref_tick (1M) as the UART source clock (SCLK), the baud rate cannot reach 921600. Better not to exceed 115200
thanks !!

AxelLin, 2020-02-26:
Hi @koobest,
Thanks.
So that means I cannot use baud_rate = 921600 and enable CONFIG_PM_ENABLE at the same time, is my understanding correct?

closed 2020-02-26 by AxelLin

Harvie, 2020-10-26:
Hello, will 115200 baud uart work with UART_SCLK_REF_TICK running at 1MHz?