espressif/esp-idf #4840: Division by zero in `esp_bt_controller_init` when `BTDM_MODEM_SLEEP` is not set

hb-kaertech, 2020-02-26:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.2-dev-404-gd6e672ed5
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Since commit d6e672ed511edd2714c42d27a89cde0afd9ec207, if `CONFIG_BTDM_MODEM_SLEEP` is not set, `esp_bt_controller_init` triggers a division by zero resulting in a crash.

### Steps to reproduce

1- Unset CONFIG_BTDM_MODEM_SLEEP in menuconfig
2- Flash&Run my code below

### Code to reproduce this issue

`main.c`:
```
#include <driver/uart.h>
#include <esp_bt.h>
#include <esp_log.h>

static const char *TAG = "app";

void app_main(void)
{
    uart_set_baudrate(0, 2000000);
    ESP_LOGI(TAG, "App starting");

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);
}
```

`sdkconfig`: https://gist.github.com/hb-kaertech/cf76a52a66d0a5322a5b8de6d51347e3

## Debug Logs

```
I (371) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (379) app: App starting
I (379) BTDM_INIT: BT controller compile version [ebdefc3]
I (379) system_api: Base MAC address is not set
I (379) system_api: read default base MAC address from EFUSE
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000d099  PS      : 0x00060f30  A0      : 0x800829f8  A1      : 0x3ffc4d00  
A2      : 0x00000064  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000064  A12     : 0x3ffc4d34  A13     : 0x3ffb8364  
A14     : 0x400e2048  A15     : 0x3ffc4cf8  SAR     : 0x00000020  EXCCAUSE: 0x00000006  
0x400e2048: r_flash_read at ??:?

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 99be06220bd84b53dc78f88a79832594da8f509d8322fb2ceda4330a706657d9

Backtrace: 0x4000d096:0x3ffc4d00 0x400829f5:0x3ffc4d20 0x4003c166:0x3ffc4d40 0x4003c3bc:0x3ffc4d60 0x40055199:0x3ffc4da0 0x400e1916:0x3ffc4dc0 0x400d456f:0x3ffc4e00 0x400d4a90:0x3ffc4e20 0x400909a1:0x3ffc4e50
0x400829f5: btdm_us_2_lpcycles at esp-idf/components/bt/controller/bt.c:806

0x400e1916: r_rwip_init at ??:?

0x400d456f: r_rw_pre_main at arch_main.c:?

0x400d4a90: btdm_controller_task at ??:?

0x400909a1: vPortTaskWrapper at esp-idf/components/freertos/xtensa/port.c:143

Rebooting...
```

hb-kaertech, 2020-02-26:
@wmy-espressif: can you have a look at this?

mywang-espressif, 2020-02-26:
HI @hb-kaertech , this is a bug introduced in a recent code merge and I am sorry for the mistake. I am fixing this issue and the patch can be expected in a couple of days.

mywang-espressif, 2020-03-02:
Hi @hb-kaertech , this issue is fixed in recent merge(commit SHA: 40baff19f83aa2f8c31645433ef7b28b77cd78ad). You can verify that.

hb-kaertech, 2020-03-02:
Hello @mywang-espressif, I confirm it fixes the issues for me. Thank you!

closed 2020-03-02 by hb-kaertech