espressif/esp-idf #4841: ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400902ec

sahilsoni1, 2020-02-26:
----------------------------- Delete below -----------------------------
I (31) boot: ESP-IDF v4.2-dev-432-gc65b671-dirty 2nd stage bootloader
I (31) boot: compile time 18:19:19
I (32) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1647c ( 91260) map
I (148) esp_image: segment 1: paddr=0x000264a4 vaddr=0x3ffb0000 size=0x03358 ( 13144) load
I (154) esp_image: segment 2: paddr=0x00029804 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/sahilsoni/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1789

I (155) esp_image: segment 3: paddr=0x00029c10 vaddr=0x40080404 size=0x06408 ( 25608) load
I (175) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x72fbc (470972) map
0x400d0020: _stext at ??:?

I (345) esp_image: segment 5: paddr=0x000a2fe4 vaddr=0x4008680c size=0x0ec40 ( 60480) load
0x4008680c: xRingbufferSend at /home/sahilsoni/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1011

I (384) boot: Loaded app from partition at offset 0x10000
I (384) boot: Disabling RNG early entropy source...
I (384) cpu_start: Pro cpu up.
I (388) cpu_start: Application information:
I (393) cpu_start: Project name:     fast_scan
I (398) cpu_start: App version:      v4.2-dev-432-gc65b671-dirty
I (405) cpu_start: Compile time:     Feb 26 2020 18:19:33
I (411) cpu_start: ELF file SHA256:  b0e0e2227eb2c5a4...
I (417) cpu_start: ESP-IDF:          v4.2-dev-432-gc65b671-dirty
I (423) cpu_start: Starting app cpu, entry point is 0x4008147c
0x4008147c: call_start_cpu1 at /home/sahilsoni/esp/esp-idf/components/esp32/cpu_start.c:274

I (415) cpu_start: App cpu up.
I (434) heap_init: Initializing. RAM available for dynamic allocation:
I (441) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (447) heap_init: At 3FFB8BA8 len 00027458 (157 KiB): DRAM
I (453) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (459) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (466) heap_init: At 4009544C len 0000ABB4 (42 KiB): IRAM
I (472) cpu_start: Pro cpu start user code
I (490) spi_flash: detected chip: gd
I (491) spi_flash: flash io: dio
W (491) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (501) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (590) net80211: OS adapter function version error! Version 4 is expected, but it is 5
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400902ec
0x400902ec: _esp_error_check_failed at /home/sahilsoni/esp/esp-idf/components/esp32/panic.c:739

igrr, 2020-02-26:
Please run `git submodule update --init` and build the program again.

sahilsoni1, 2020-02-28:
Thank u, now it is working.

closed 2020-02-28 by igrr