espressif/esp-idf #4845: Build error when console UART set to None

KaeLL, 2020-02-26:
IDF git describe: v4.2-dev-432-gc65b67115
```
esp-idf/components/console/esp_console_repl.c:125:22: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared
```
The existence of this constant is tied to the default setting, but it's still used in IDF.

AxelLin, 2020-03-02:
I got the same build error with v4.2-dev-521-g6fd855ab8d00.

AxelLin, 2020-03-05:
Hi @Alvin1Zhang @suda-morris 

Any chance to fix the build error ASAP? IMHO, a build breakage should be high priority to fix.

BTW, I think esp_console_repl_init() should also takes "const uart_config_t *uart_config" argument
so the caller can use different uart settings.

suda-morris, 2020-03-05:
@KaeLL Sorry for the delay, we will fix it ASAP. Thanks for your nice suggestion. @AxelLin

closed 2020-03-10 by espressif-bot