espressif/esp-idf #4847: fatal: No names found, cannot describe anything

PauliNioh, 2020-02-27:
2020-02-27 09:42:55.991   Running command: C:\Users\Paul\AppData\Local\Programs\Python\Python37\python.exe C:\Users\Paul\Desktop\esp32\esp-idf-master\tools\idf_tools.py install-python-env
                          fatal: No names found, cannot describe anything
                          
.
.
.

   WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128.

[image]

Alvin1Zhang, 2020-02-27:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

dobairoland, 2020-02-27:
@PauliNioh How the output continues? I assume that everything works as it should be and you get to "Done! You can now compile ESP-IDF projects. Go to the project directory and run: idf.py build". 

If your ESP-IDF is a shallow clone then `git describe --tags` fails but `idf_tools.py` detects the IDF version in a different way.

closed 2020-02-27 by PauliNioh