espressif/esp-idf #4848: 国内clone中途失败

yhongm, 2020-02-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

sheldon123z, 2020-02-27:
我觉得使用git 克隆github上的在国内也一样用吧...?

suda-morris, 2020-02-28:
Hi @yhongm 国内访问GitHub慢是一个众所周知的事情，原因相信你一定了解...经过实践发现，如果你连接 4G 热点然后再clone仓库，速度会快上很多，你可以试试看。另外还有一个办法，就是在我们的release页面下载zip源码包(他们位于dl.espressif.com，所以下载速度会很快），解压后可以通过`git fetch`等命令将master分支上的最新提交同步下来（这样你从github上直接clone的数据量会少很多很多）。

我们不使用gitee的一个重要原因是，ESP-IDF中包含了许多**第三方**的子模块（通过 git submodule 进行管理），他们中的绝大部分也都在Github上托管着，我们无法要求这些项目的作者也去Gitee上创建一个镜像仓库，你说对吧？

redchenjs, 2020-02-28:
确实，即使主仓库托管在gitee上，那些子模块还是会链接到github上，所以也差不了多少

yhongm, 2020-02-28:
> Hi @yhongm 国内访问GitHub慢是一个众所周知的事情，原因相信你一定了解...经过实践发现，如果你连接 4G 热点然后再clone仓库，速度会快上很多，你可以试试看。另外还有一个办法，就是在我们的release页面下载zip源码包(他们位于dl.espressif.com，所以下载速度会很快），解压后可以通过`git [...]

好吧，谢谢

yhongm, 2020-02-28:
clone了一个星期了，无论是移动4G，电信4G，宽带。每次clone百分之十几就中途失败。一个星期没干别的，就clone这个了。只能说无语两个字

yhongm, 2020-02-28:
[image]
[image]
我用了github几年，从来没有clone一个项目这么费劲，clone一个星期，无数次失败。

yhongm, 2020-02-28:
[image]

suda-morris, 2020-02-28:
很抱歉因为网络的原因带来不愉快的使用体验 😭

从 release 页面直接下载 zip 包呢，是否可以成功？

另外，国内也有一些提供Github下载加速的网站，比如这个：http://g.widora.io/

yhongm, 2020-02-28:
> 很抱歉因为网络的原因带来不愉快的使用体验 😭
> 
> 从 release 页面直接下载 zip 包呢，是否可以成功？
> 
> 另外，国内也有一些提供Github下载加速的网站，比如这个：http://g.widora.io/

> 很抱歉因为网络的原因带来不愉快的使用体验 😭
> 
> 从 release 页面直接下载 zip 包呢，是否可以成功？
> 
> 另外，国内也有一些提供Github下载加速的网站，比如这个：http://g.widora.io/

谢谢

closed 2020-02-28 by yhongm