espressif/esp-idf #4850: MCPWM example does not work correctly (only 2 PWM signals on 6 possible outputs)

salocinx, 2020-02-27:
## Environment

- Development Kit:        ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROVER
- IDF version:                v4.2-dev-414-g132cc67c0
- Build System:             CMake
- Compiler version:       xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux (Ubuntu 18.04 LTS)
- Using an IDE?:           Eclipse CDT 2019.12
- Power Supply:            external 5V / 5A

## Problem Description

The MCPWM Basic Config example does not work correctly. I didn't change anything, just tried to run the [mcpwm_basic_config_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c) as it is on GitHub.

### Expected Behavior

The [mcpwm_basic_config_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c) should output a PWM signal on all six ouputs `PWM0A`, `PWM0B`, `PWM1A`, `PWM1B`, `PWM2A` and `PWM2B`.

### Actual Behavior

PWM signal does only on `PWM0A` and `PWM0B` GPIOs exist. There's **no** PWM signal at `PWM1A`, `PWM1B`, `PWM2A` and `PWM2B` GPIOs.

### Steps to reproduce

1. Download [mcpwm_basic_config_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c)
2. Flash to ESP32 Wrover Development Kit
3. Measure PWM signals on defined pins with oscilloscope or by using a DC motor with an H-Bridge

### Code to reproduce this issue

[mcpwm_basic_config_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/mcpwm/mcpwm_basic_config/main/mcpwm_basic_config_example.c)

## Debug Logs

No error messages. Let me know if I can provide any logs.

negativekelvin, 2020-02-27:
Gpio 16/17 are not available on wrover. Check the datasheet.

salocinx, 2020-02-27:
Thank you! That solved my issue. I am going to close this one (if I can).

It's really nasty, that these GPIOs are broken out on my ESP Wrover Dev Kit. Very misleading and frustrating.

closed 2020-02-27 by salocinx