espressif/esp-idf #4851: ble_mesh_server/client example failed

sheldon123z, 2020-02-27:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.0

- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Promp].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Hi, I am using ESP32 DevKitC to test the ble_mesh_server/client examples, everything looks like going well after I flashed the example into the board. However, I couldn't use my phone to detect the board in either the nRF Mesh application or the ESP-BLE-Prov application. When I use the cmd to monitor the board status, I found the board is rebooting all the time, and the log information is like below:
 
## Debug Logs

```
C:\Users\zheng\Desktop\esp-idf\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server>idf.py build
Executing action: all (aliases: build)
Running cmake in directory c:\users\zheng\desktop\esp-idf\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 c:\users\zheng\desktop\esp-idf\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.25.1.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/zheng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/zheng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/zheng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/zheng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/zheng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Loading defaults file C:/Users/zheng/Desktop/esp-idf/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/sdkconfig.defaults...
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs-time.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Found PythonInterp: C:/Users/zheng/.espressif/python_env/idf4.2_py3.6_env/Scripts/python.exe (found version "3.6.5")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "onoff_server" version: v4.2-dev-459-ge36516372
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/zheng/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/zheng/Desktop/esp-idf/components/app_trace C:/Users/zheng/Desktop/esp-idf/components/app_update C:/Users/zheng/Desktop/esp-idf/components/asio C:/Users/zheng/Desktop/esp-idf/components/bootloader C:/Users/zheng/Desktop/esp-idf/components/bootloader_support C:/Users/zheng/Desktop/esp-idf/components/bt C:/Users/zheng/Desktop/esp-idf/components/cbor C:/Users/zheng/Desktop/esp-idf/components/coap C:/Users/zheng/Desktop/esp-idf/components/console C:/Users/zheng/Desktop/esp-idf/components/cxx C:/Users/zheng/Desktop/esp-idf/components/driver C:/Users/zheng/Desktop/esp-idf/components/efuse C:/Users/zheng/Desktop/esp-idf/components/esp-tls C:/Users/zheng/Desktop/esp-idf/components/esp32 C:/Users/zheng/Desktop/esp-idf/components/esp_adc_cal C:/Users/zheng/Desktop/esp-idf/components/esp_common C:/Users/zheng/Desktop/esp-idf/components/esp_eth C:/Users/zheng/Desktop/esp-idf/components/esp_event C:/Users/zheng/Desktop/esp-idf/components/esp_gdbstub C:/Users/zheng/Desktop/esp-idf/components/esp_http_client C:/Users/zheng/Desktop/esp-idf/components/esp_http_server C:/Users/zheng/Desktop/esp-idf/components/esp_https_ota C:/Users/zheng/Desktop/esp-idf/components/esp_https_server C:/Users/zheng/Desktop/esp-idf/components/esp_local_ctrl C:/Users/zheng/Desktop/esp-idf/components/esp_netif C:/Users/zheng/Desktop/esp-idf/components/esp_ringbuf C:/Users/zheng/Desktop/esp-idf/components/esp_rom C:/Users/zheng/Desktop/esp-idf/components/esp_serial_slave_link C:/Users/zheng/Desktop/esp-idf/components/esp_timer C:/Users/zheng/Desktop/esp-idf/components/esp_websocket_client C:/Users/zheng/Desktop/esp-idf/components/esp_wifi C:/Users/zheng/Desktop/esp-idf/components/espcoredump C:/Users/zheng/Desktop/esp-idf/components/esptool_py C:/Users/zheng/Desktop/esp-idf/components/expat C:/Users/zheng/Desktop/esp-idf/components/fatfs C:/Users/zheng/Desktop/esp-idf/components/freemodbus C:/Users/zheng/Desktop/esp-idf/components/freertos C:/Users/zheng/Desktop/esp-idf/components/heap C:/Users/zheng/Desktop/esp-idf/components/idf_test C:/Users/zheng/Desktop/esp-idf/components/jsmn C:/Users/zheng/Desktop/esp-idf/components/json C:/Users/zheng/Desktop/esp-idf/components/libsodium C:/Users/zheng/Desktop/esp-idf/components/log C:/Users/zheng/Desktop/esp-idf/components/lwip C:/Users/zheng/Desktop/esp-idf/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main C:/Users/zheng/Desktop/esp-idf/components/mbedtls C:/Users/zheng/Desktop/esp-idf/components/mdns C:/Users/zheng/Desktop/esp-idf/components/mqtt C:/Users/zheng/Desktop/esp-idf/components/newlib C:/Users/zheng/Desktop/esp-idf/components/nghttp C:/Users/zheng/Desktop/esp-idf/components/nvs_flash C:/Users/zheng/Desktop/esp-idf/components/openssl C:/Users/zheng/Desktop/esp-idf/components/partition_table C:/Users/zheng/Desktop/esp-idf/components/perfmon C:/Users/zheng/Desktop/esp-idf/components/protobuf-c C:/Users/zheng/Desktop/esp-idf/components/protocomm C:/Users/zheng/Desktop/esp-idf/components/pthread C:/Users/zheng/Desktop/esp-idf/components/sdmmc C:/Users/zheng/Desktop/esp-idf/components/soc C:/Users/zheng/Desktop/esp-idf/components/spi_flash C:/Users/zheng/Desktop/esp-idf/components/spiffs C:/Users/zheng/Desktop/esp-idf/components/tcp_transport C:/Users/zheng/Desktop/esp-idf/components/tcpip_adapter C:/Users/zheng/Desktop/esp-idf/components/ulp C:/Users/zheng/Desktop/esp-idf/components/unity C:/Users/zheng/Desktop/esp-idf/components/vfs C:/Users/zheng/Desktop/esp-idf/components/wear_levelling C:/Users/zheng/Desktop/esp-idf/components/wifi_provisioning C:/Users/zheng/Desktop/esp-idf/components/wpa_supplicant C:/Users/zheng/Desktop/esp-idf/components/xtensa
-- Configuring done
-- Generating done
[... cut ...]
Backtrace: 0x4000d096:0x3ffb5d00 0x40082975:0x3ffb5d20 0x4003c166:0x3ffb5d40 0x4003c3bc:0x3ffb5d60 0x4013e862:0x3ffb5da0 0x40055199:0x3ffb5dc0 0x40147092:0x3ffb5de0 0x401399e7:0x3ffb5e20 0x40139f24:0x3ffb5e40 0x400902d1:0x3ffb5e70
0x40082975: btdm_us_2_lpcycles at C:/Users/zheng/Desktop/esp-idf/components/bt/controller/bt.c:806

0x4013e862: r_ld_init_hark at ??:?

0x40147092: r_rwip_init at ??:?

0x401399e7: r_rw_pre_main at arch_main.c:?

0x40139f24: btdm_controller_task at ??:?

0x400902d1: vPortTaskWrapper at C:/Users/zheng/Desktop/esp-idf/components/freertos/xtensa/port.c:143

Rebooting...

```
Any thought about my problem?

Campou, 2020-03-02:
Hi @sheldon123z 

Sorry for the inconvenience caused.

Currently please enable the "Bluetooth modem sleep" in the bluetooth controller menuconfig, this will fix this issue. And we have fixed this issue internally, the changes will be synchronized to Github very soon.

Thanks and Regards

Campou, 2020-03-03:
Hi @sheldon123z 

The branch [release/v4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) has been updated, please pull and have a try.

Thanks.

sheldon123z, 2020-03-04:
Hi @Campou ,

Thank you very much, that works!

I also have a quick question, since I am new to BLE mesh, I am still learning the stack. I found the examples seem not able to be provisioned by other provisioners except for the phone and the example provisioner, like an embedded Linux provisioner Raspberry Pi3 B+ using BlueZ, and the Tmall genie, which also uses Bluetooth mesh as the communication protocol. Both of them are not working with the examples. What adjustment should I do to handle this?

Thanks!

closed 2020-03-12 by sheldon123z