espressif/esp-idf #4855: UART STOPS READING WHEN FOUND THE BYTE 0x00

romocitto88, 2020-02-28:
## Environment

- Development Kit:      ESP32-Dev-Kit
- Module or chip used:  ESP32-WROOM-32D
- IDF version latest stable version in PlatformIO
- Build System: idf.py
- Operating System: Windows
- Using an IDE?: Yes PlatformIO
- Power Supply: USB

## Problem Description

I'm using the ble server and client uart bridge example to send my frame from side A to side B and viceversa.
I found that the reading of Uart bytes stops when found a 0x00, so the frame is incomplete.
Anyone know how to solve it?

### Code to reproduce this issue

```cpp

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "driver/uart.h"

#include "esp_bt.h"
#include "nvs_flash.h"
#include "esp_bt_device.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"
#include "esp_system.h"
#include "esp_gatt_common_api.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"

#define GATTC_TAG                   "GATTC_SPP_DEMO"
#define PROFILE_NUM                 1
#define PROFILE_APP_ID              0
#define BT_BD_ADDR_STR              "%02x:%02x:%02x:%02x:%02x:%02x"
#define BT_BD_ADDR_HEX(addr)        addr[0],addr[1],addr[2],addr[3],addr[4],addr[5]
#define ESP_GATT_SPP_SERVICE_UUID   0xABF0
#define SCAN_ALL_THE_TIME           0

struct gattc_profile_inst {
    esp_gattc_cb_t gattc_cb;
    uint16_t gattc_if;
    uint16_t app_id;
    uint16_t conn_id;
    uint16_t service_start_handle;
    uint16_t service_end_handle;
    uint16_t char_handle;
    esp_bd_addr_t remote_bda;
};

enum{
    SPP_IDX_SVC,

    SPP_IDX_SPP_DATA_RECV_VAL,

    SPP_IDX_SPP_DATA_NTY_VAL,
    SPP_IDX_SPP_DATA_NTF_CFG,

    SPP_IDX_SPP_COMMAND_VAL,

    SPP_IDX_SPP_STATUS_VAL,
    SPP_IDX_SPP_STATUS_CFG,

[... 556 lines cut ...]
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ESP_LOGI(GATTC_TAG, "%s init bluetooth\n", __func__);
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
 
    ble_client_appRegister();
    spp_uart_init();
}

```

macdroid53, 2020-02-28:
Way too much code to wade through, but I'll take a _guess_ that it's the issue that started, or is related to, this issue:
https://github.com/espressif/esp-idf/issues/4537

romocitto88, 2020-02-28:
This issue is something different: when a byte with value 0x00 is reached from Usart causes a stop of reading.
For example if my frame is: "0x01 0x02 0x03 0x00 0x01 0x03", when I exec:
```cpp
[..]
uart_read_bytes(UART_NUM_1,temp,event.size,portMAX_DELAY);
[..]
```

The function writes in my buffer "temp" only: {0x01, 0x02, 0x03} 
So it's truncated.

macdroid53, 2020-02-28:
Hmm...if the 0x00 is being seen by the uart as a break character and code higher up flushes the buffer on break detect? Again, just guessing.

romocitto88, 2020-02-28:
The event queue even_type never equals to UART_BREAK, so i presume that the break isn't decteted.

macdroid53, 2020-02-28:
Ok, so that seems to rule out break detect issues.

Is there a long enough time between 0x03 and 0x00 to trigger a timeout?

romocitto88, 2020-03-03:
I found the problem! And it's a very stupid error. This issue can be closed.

I'm sorry.

closed 2020-03-03 by romocitto88

Yuxekova, 2021-05-09:
I am facing same problem, it would be good if you post the solution. IDK why uart stops when founds 0x00.

romocitto88, 2021-05-10:
In function 
>void uart_task (void * pvParameters)

 I was showing, in the serial debug, the UART frame whit this line:

>ESP_LOGE (GATTC_TAG," WR: ---% s --- \ n ", temp);

But it is wrong, because the "temp" buffer, in my case, was not a string but a sequence of byte. So, when 0x00 value was found for the first time, the frame was resulting truncated.