espressif/esp-idf #4856: esp_timer_create is not acurate when using power management

JosuGZ, 2020-02-28:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- (Windows only) environment type: MSYS2 mingw32

## Problem Description

I'm using a timer to send a LoRa packet to a receiver, both being ESP32 (same board).

On the receiver, I measure the time between packets by using `esp_timer_get_time`, and when both are running at 80Mhz, the time is correct: when sending every 10000ms, I receive every 10000ms.

The problem arises when I enable power management on the sender: if I use sleep *or* frequency scaling, the packets arrive *sooner* than expected, for example about 40ms sooner when sending every 10000ms and using light sleep.

Example log with light sleep:

```
I (8642320) rfm95: Time since last packet: 9961
I (8652281) rfm95: Time since last packet: 9961
I (8662242) rfm95: Time since last packet: 9960
```

Example log with no power management:

```
I (8822482) rfm95: Time since last packet: 10000
I (8832482) rfm95: Time since last packet: 10000
I (8842482) rfm95: Time since last packet: 10000
I (8852482) rfm95: Time since last packet: 10000

```

Example log with frequency scaling (min: 40, max: 80):

```
I (21370) rfm95: Time since last packet: 9944
I (31314) rfm95: Time since last packet: 9944
I (41261) rfm95: Time since last packet: 9947
```

### Expected Behavior

A periodic timer configured to 10000ms should run every 10000ms or more.

### Actual Behavior

The timer runs faster than 10000ms when power management is enabled.

igrr, 2020-02-28:
Hi @JosuGZ, which time source do you use in the callback to print the elapsed time between callbacks? Is it also esp_timer or something else?

JosuGZ, 2020-02-28:
It is `esp_timer_get_time`, it is on the receiver, which is always at a fixed frequency of 80Mhz (same codebase and same sdkonfig, no pm enable call).

JosuGZ, 2020-02-29:
Keep in mind that when I measure on the sender, I always get 10000ms (+/-1ms). The difference appears on the receiver when the receiver is running at a fixed frequency and the sender has power management enabled.

JosuGZ, 2020-03-02:
It seems like I get much better accuracy if I change the `RTC clock source` from the `internal 150kHz RC oscillator` to the `internal 8.5Mhz oscillator, divided by 256 (~33kHz)`, but I don't know the implications of this.

```diff
-CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_RC=y
+CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_RC=
-CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256=
+CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_8MD256=y
```

JosuGZ, 2020-03-02:
If I put a min frequency of 10mHz it gets even worse (the expected time is 10000ms):

```
I (5704687) rfm95: Time since last packet: 7657
I (5712074) rfm95: Time since last packet: 7387
I (5719413) rfm95: Time since last packet: 7339
I (5726770) rfm95: Time since last packet: 7357
```

xiaojie76, 2022-05-23:
Hi@JosuGZ  The esp_timer used by esp32 is implemented by the timer_group peripheral, which uses APB_CLK, so after using the dynamic frequency scaling of power management, the timer count will change with the change of the APB frequency, resulting in inaccurate timing . So it is recommended not to change APB_CLK when using esp_timer. There are two usage suggestions:
1. Use PM_LOCK
```
esp_pm_lock_create(ESP_PM_APB_FREQ_MAX, ....);
esp_pm_lock_acquire();
```
2. `pm_config.min_freq_mhz = 80MHz`

In addition, other chips will not have this issue because esp_timer uses the systimer peripheral, and even if the APB_CLK clock source is switched, the systimer will perform automatic time compensation to improve timer accuracy.

closed 2023-03-17 by Alvin1Zhang