espressif/esp-idf #4857: how to link the ip address or tcp method got from my gsm module to esp module . and use default mqtt library to send data I dont want to use ppos method . becuase i dont have knowledge exposure on ppos method .

manojsai17, 2020-02-28:
we interface the gsm module to esp uart .
I able to data transfer to cloud using wifi with default mqtt library .
So we interfaced gsm module to esp uart .
with traditional at command i able to get the network and ip address .

how to link the ip address or tcp method got from my gsm module to esp module . and use default mqtt library to send data to cloud.
I dont want to use ppos method . becuase i dont have knowledge exposure on ppos method .

negativekelvin, 2020-02-29:
1. Do not post duplicate questions here and on the forum.  Only post bug reports here, not questions.

2. You have 9 open issues.  Please close anything that is not a bug report.

3. If you need urgent assistance then email sales@espressif.com and request technical support through your account representative.

Alvin1Zhang, 2020-03-02:
@negativekelvin Thanks for the notes. @manojsai17 Thanks for reporting. For how questions and third party questions, it is suggested that you could raise a post on our forum at https://esp32.com, thanks. If you have requests about feature, issue and etc, please help follow template [here](https://github.com/espressif/esp-idf/issues/new/choose). Thanks.

closed 2020-03-02 by Alvin1Zhang