espressif/esp-idf #4860: 'cmake' must be available on the PATH to use /Users/copy/esp/esp-idf/tools/idf.py

xiaoxinxing12, 2020-02-29:
bogon:hello_world copy$ idf.py menuconfig
'cmake' must be available on the PATH to use /Users/copy/esp/esp-idf/tools/idf.py
bogon:hello_world copy$

bitmakk, 2020-03-01:
Yes, it is needed to function properly, as mentioned in docs.
Add the cmake binary to the PATH environment variable.

dobairoland, 2020-03-02:
@xiaoxinxing12 Please tell us what operating system are you using (and its version & terminal) and what is the version of your IDF in order to give you a better recommendation. The general recommendation is that you need to install `cmake` and make it available so running `cmake --version` would be possible from your terminal window.

dobairoland, 2020-03-02:
You may have MacOs judging from the path in the title. Cmake installation is discussed here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html#install-prerequisites

dobairoland, 2020-03-19:
I'm closing this for lack of activity.

closed 2020-03-19 by dobairoland

ingconti, 2021-03-22:
go plain:
a) download CMAKE App fro its site
b) it will be in:   /Applications/CMake.app/Contents/bin 
c) add to your shell using nano to edit . zprofile:
    nano  ~/.zprofile  
d) in nano add:
   
      #settinng path for cmake for expressif
     export PATH=$PATH:/Applications/CMake.app/Contents/bin/

Th3Et3rnalz, 2021-09-02:
Another, probably obvious thing to check (but I did forget, so here I am) is that you actually have cmake installed. Running 'whereis cmake' gave some valid paths, so I thought it was installed. Installing cmake through my standard package manager fixed the problem for me.