espressif/esp-idf #4861: the program gets stuck after getting "Got IP Event"

sumit809, 2020-02-29:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-PICO-Kit]
- Kit version (PicoKit): [v4]
- Module or chip used:  [ESP32-PICO-D4|ESP32]
- IDF version: v4.2-dev-459-ge36516372-dirty
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

the program gets stuck after getting "Got IP Event"

### Expected Behavior
It should send data over UDP

### Code to reproduce this issue

```

/* 
UART Interrupt Example
*/
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/queue.h"
#include "driver/uart.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "esp_intr_alloc.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
// #include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/soc.h"
#include <stdarg.h>
[... 430 lines cut ...]
I (662) wifi: Init management short buffer num: 32
I (662) wifi: Init static rx buffer size: 1600
I (672) wifi: Init static rx buffer num: 10
I (672) wifi: Init dynamic rx buffer num: 32
[0;32mI (682) example_connect: Connecting to ADAPT...[0m
[0;32mI (772) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
I (772) wifi: mode : sta (d8:a0:1d:40:70:60)
[0;32mI (772) example_connect: Waiting for IP[0m
I (902) wifi: new:<3,1>, old:<1,0>, ap:<255,255>, sta:<3,1>, prof:1
I (902) wifi: state: init -> auth (b0)
I (902) wifi: state: auth -> assoc (0)
I (912) wifi: state: assoc -> run (10)
I (1022) wifi: connected with ADAPT, aid = 2, channel 3, 40U, bssid = 70:4f:57:02:62:0f
I (1022) wifi: security type: 3, phy: bgn, rssi: -65
I (1022) wifi: pm start, type: 1

I (1042) wifi: AP's beacon interval = 102400 us, DTIM period = 1
[0;32mI (2612) esp_netif_handlers: sta ip: 192.168.1.114, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (2612) example_connect: Got IP event![0m

```

## Other items if possible

I have changed SSID, password, dns and port for udp in menuconfig

sumit809, 2020-03-03:
Can anybody please help me here?

Alvin1Zhang, 2020-03-03:
@sumit809 Thanks for reporting, we will look into, thanks.

david-cermak, 2020-03-03:
@sumit809 I can see that the ESP received an IPv4 event and keeps waiting (probably for the IPv6 address). Could you please check if the IPv6 is supported in your local network and if not, [this option](https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/Kconfig.projbuild#L177)  is off? Could you also try to disable the `EXAMPLE_CONNECT_IPV6` to see if the issue goes away?

sumit809, 2020-03-03:
hi @david-cermak my local network does not support IPv6. Also, I'm new to ESP-IDF, can you tell me how to disable '''EXAMPLE_CONNECT_IPV6''' in that particular? Do you mean to say commenting? If yes, then using  '''//''' ?

david-cermak, 2020-03-03:
@sumit809 This is the menuconfig, please read the documentation of project configuration https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#project-configuration-menu.

Assume you've already used that as saying
> I have changed SSID, password, dns and port for udp in menuconfig

sumit809, 2020-03-03:
@david-cermak I had changed SSID in the beginning by literally editing the file as I didn't know about this menuconfig. Anyway the issue is resolved by disabling the IPv6. Thanks :)

closed 2020-03-03 by sumit809