espressif/esp-idf #4862: cpu_start chrashes when CONFIG_SECURE_FLASH_ENC_ENABLED and LOG_LEVEL>ERROR

jas39, 2020-02-29:
cpu_start calls esp_efuse_api which uses ESP_LOG* instead of ESP_EARLY_LOG_*

- [ ] coredump as follows
```
0x4008229d: esp_log_write at C:/Users/A/esp-idf/v4.0/components/log/log.c:196

0x400e5416: esp_efuse_get_coding_scheme at C:/Users/A/esp-idf/v4.0/components/efuse/src/esp_efuse_api.c:200 (discriminator 6)

0x400e5649: check_range_of_bits at C:/Users/A/esp-idf/v4.0/components/efuse/src/esp_efuse_utility.c:443

0x400e56e6: esp_efuse_utility_process at C:/Users/A/esp-idf/v4.0/components/efuse/src/esp_efuse_utility.c:87

0x400e5319: esp_efuse_read_field_blob at C:/Users/A/esp-idf/v4.0/components/efuse/src/esp_efuse_api.c:45

0x400e6436: esp_get_flash_encryption_mode at C:/Users/A/esp-idf/v4.0/components/bootloader_support/src/flash_encrypt.c:79

0x400e648f: esp_flash_encryption_init_checks at C:/Users/A/esp-idf/v4.0/components/bootloader_support/src/flash_encrypt.c:48

0x40081548: call_start_cpu0 at C:/Users/A/esp-idf/v4.0/components/esp32/cpu_start.c:209 (discriminator 1)
```

Alvin1Zhang, 2020-03-02:
@jas39 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

KonstantinKondrashov, 2020-03-11:
Hi @jas39 !
Thanks for your report. I will fix this on the master and will backport to the previous version as well.

closed 2020-03-19 by espressif-bot