espressif/esp-idf #4863: HTTPD get client ip

raldone01, 2020-02-29:
I need to get the clients ip. I have read through https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_server.html but I couldn't find a solution.

mahavirj, 2020-03-02:
Following API can be helpful:

https://github.com/espressif/esp-idf/blob/6fd855ab8d00d23bad4660216bc2122c2285d5be/components/esp_netif/include/esp_netif_sta_list.h#L51

Please prefer esp32.com forum for general questions.

raldone01, 2020-03-02:
This only returns the ips of the connected devices not the device making the request to the webserver.

shubhamkulkarni97, 2020-03-04:
@raldone01 Please try using following code.

```
esp_err_t get_remote_ip(httpd_req_t *req, struct sockaddr_in6* addr_in)
{
    int s = httpd_req_to_sockfd(req);
    socklen_t addrlen = sizeof(*addr_in);
    if (lwip_getpeername(s, (struct sockaddr *)addr_in, &addrlen) != -1) {
        return ESP_OK;
    } else {
        ESP_LOGE(TAG, "Error getting peer's IP/port");
        return ESP_FAIL;
    }
}
```

You can call this function from URI handler. For example,

```
struct sockaddr_in6 addr_in;
    if (get_remote_ip(req, &addr_in) == ESP_OK) {
        ESP_LOGI(TAG, "Rempte IP is %s", inet_ntoa(addr_in.sin6_addr.un.u32_addr[3]));
    }
```

closed 2020-03-04 by raldone01

dowonderatwill, 2021-10-05:
I was looking for it and it works. Thank you.