espressif/esp-idf #4864: espcoredump is leaking filesystem space

amotl, 2020-02-29:
Dear Espressif team,

I know this is a rather uncommon report. It's not about ESP-IDF itself but about the [`espcoredump.py`](https://github.com/espressif/esp-idf/blob/master/components/espcoredump/espcoredump.py) program shipped with the ESP-IDF.

We are using it to investigate several core dumps we are getting from our devices and found it would leak large amounts of filesystem memory on our workstation. Despite trying very hard already and looking at the temporary files created by `espcoredump.py`, we haven't been able to find the root cause and the place where those files would be stored on the filesystem.

Apparently, it is not `/tmp`, `/var/folders/nw/f_dhlwbn5rl7x4wlzq3mhw0m0000gp/T` or any other location where temporary files are usually placed within. We are running macOS.

As the amounts of storage leaked is in the gigabytes, we've exhausted the filesystem storage on our workbench after a few invocations and are not able to get it back somehow.

Thanks already for taking the time to look into this.

With kind regards,
Andreas.

amotl, 2020-03-01:
As a side note, we have been able to reclaim some disk space by shrinking the `Docker.raw` file on our machine, see also https://github.com/docker/for-mac/issues/2297 in order to continue investigating this.

However, the `df -hm` output before running one invocation of `espcoredump` is 62956 and after it, it yields only `58730`. This is a difference of 4,226 MB (4 GB) and we still have no idea where this is going to.

amotl, 2020-03-01:
We have been able to spot the filesystem location where the some core files have been dropped on our system. It is the `/cores` folder which accumulated some gigabytes worth of memory.
```
$ du -sch /cores
159G	/cores
```

A single invocation of `espcoredump` would yield three `core.XXXX` files worth of 3.8G.
```
$ du -sch /cores
3.8G	/cores
```

amotl, 2020-03-01:
## Root cause

It turns out, the naming scheme/pattern of the core dump file(s) on Linux is determined through
```
$ cat /proc/sys/kernel/core_pattern
core
```

However, on macOS, the default is
```
$ sysctl kern.corefile
kern.corefile: /cores/core.%P
```

When setting this to
```
$ sudo sysctl -w kern.corefile=core
```

1. only a single core dump file will be created within the working directory and
2. the overall process will be much faster as it would apparently not create three core files.

## References
- https://apple.stackexchange.com/questions/108861/can-os-x-dump-cores-in-the-working-directory-instead-of-cores
- http://www.brendangregg.com/blog/2016-08-09/gdb-example-ncurses.html

closed 2020-03-02 by amotl

gerekon, 2020-03-03:
Hmm, it seems to be core files of crashed GDB. `espcoredump.py` restarts it when it crashes to continue printing the rest of the info. Xtensa GDB can crash during parsing the erroneous stack. 
What version of GDB do you use?

amotl, 2020-03-04:
Dear Alexey,

indeed, the crashing GDB could be the real reason for that as it reports respective things to me while running `espcoredump`. I am still on xtensa-gcc 5.2 something.

With kind regards,
Andreas.