espressif/esp-idf #4869: idf monitor

PauliNioh, 2020-03-01:
[image]

Alvin1Zhang, 2020-03-02:
Thanks for reporting. Would you please help provide more details as suggested in the [feature template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Feature+Request&template=feature_request.md&title=)? Thanks.

dobairoland, 2020-03-02:
@PauliNioh Do you experience any issue here? That is just a warning and the monitor should work as expected.

Note: Please ask questions on https://esp32.com/ instead of opening a Github issue for everything. This is not a feature request nor an issue.

closed 2020-03-02 by PauliNioh