espressif/esp-idf #4870: Place on the page "api-guides/build-system-legacy.html" an example file CMakeLists.txt with the connection code of several components from idf

komandor2, 2020-03-02:
I think it will be useful for many to place an example of connecting components in a CMakeLists.txt file, since for two weeks I have been unsuccessfully trying to do this, I read a lot of documentation but still have not solved this problem. I would really appreciate a code example CMakeLists.txt!

Думаю будет многим полезно разместить пример подключения компонентов в CMakeLists.txt файле, так как сам уже две недели безуспешно пытаюсь это сделать, прочитал уйму документации но так и не решил эту проблему. Буду очень признателен за пример кода!

krzychb, 2020-03-03:
Hi @komandor2, 

The page [api-guides/build-system-legacy.html](https://docs.espressif.com/projects/esp-idf/en/v4.0/api-guides/build-system-legacy.html) refers to Legacy GNU Make and is not using CMakeLists.txt.

Do you compile your projects using GNU Make  or rather meant CMake?

komandor2, 2020-03-03:
Thathks krzychb, but but I read the warning on the documentation page 
"Note
Since ESP-IDF V4.0, the default build system is based on CMake. This documentation is for the legacy build system based on GNU Make. Support for this build system may be removed in future major releases."
and decided it was better to start working with CMake.

komandor2, 2020-03-03:
it would be nice if there was an example project with several components connected. For example, from ADC, server and WPS

komandor2, 2020-03-03:
In my project, I created a file CMakeLists.txt :

cmake_minimum_required(VERSION 3.5)
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components
						$ENV{IDF_PATH}/examples/protocols/https_server/main
						$ENV{IDF_PATH}/components/esp_https_server
						$ENV{IDF_PATH}/examples/wifi/wps/main)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(asio_chat_server)

, but I'm not sure of its correctness, since there are a lot of warnings and errors during the assembly :-(

krzychb, 2020-03-03:
> it would be nice if there was an example project with several components connected. For example, from ADC, server and WPS

I agree, I see many questions how to add custom components to projects and CMakeLists.txt sample files would be helpful. 

> $ENV{IDF_PATH}/components/esp_https_server

But to be clear, components from `{IDF_PATH}/components` should be visible in your projects without adding them to any CMakeLists.txt. 

Are you able to compile unmodified examples:
- examples/peripherals/adc/
- examples/protocols/https_server/
- examples/wifi/wps/

or you have some errors?

Check CMakeLists.txt files for these projects. They do not contain any reference to any components in `{IDF_PATH}/components`.

On the other hand example `examples/protocols/https_server/` has one reference to custom component `set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)` and shows how to add custom components.

renzbagaporo, 2020-03-04:
@komandor2, adding to what @krzychb said,  two `main` components are also included, which,

1) Since component name is based on directory name, will not be unique. One will probably override the other.
2) Each probably has an `app_main` definition leading to duplicate symbol errors.

I would suggest picking one example project that most closely resembles what you want to do as a base, then copying  code from other examples as necessary.

komandor2, 2020-03-04:
if I remove component directories from a CMakelists file, then a lot of errors appear during compilation.

Example my CMakelists:
cmake_minimum_required(VERSION 3.5)
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
#						$ENV{IDF_PATH}/examples/protocols/https_server/main
#						$ENV{IDF_PATH}/components/esp_https_server
#						$ENV{IDF_PATH}/examples/wifi/wps/main)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(asio_chat_server)

Example compiling:
Building in: C:\Users\Home\eclipse-workspace\asio1\build
cmake --build . -- -v
[1/9] cmd.exe /C "cd /D C:\Users\Home\eclipse-workspace\asio1\build\bootloader && C:\Users\Home\.espressif\tools\cmake\3.13.4\bin\cmake.exe --build ."
ninja: no work to do.
[2/7] C:\Users\Home\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Users/Home/Desktop/esp-idf/components/newlib/platform_include -IC:/Users/Home/Desktop/esp-idf/components/freertos/include -IC:/Users/Home/Desktop/esp-idf/components/heap/include -IC:/Users/Home/Desktop/esp-idf/components/log/include -IC:/Users/Home/Desktop/esp-idf/components/soc/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/soc/include -IC:/Users/Home/Desktop/esp-idf/components/esp_rom/include -IC:/Users/Home/Desktop/esp-idf/components/esp_common/include -IC:/Users/Home/Desktop/esp-idf/components/xtensa/include -IC:/Users/Home/Desktop/esp-idf/components/xtensa/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/driver/include -IC:/Users/Home/Desktop/esp-idf/components/esp_ringbuf/include -IC:/Users/Home/Desktop/esp-idf/components/esp_event/include -IC:/Users/Home/Desktop/esp-idf/components/tcpip_adapter/include -IC:/Users/Home/Desktop/esp-idf/components/lwip/include/apps -IC:/Users/Home/Desktop/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Home/Desktop/esp-idf/components/lwip/lwip/src/include -IC:/Users/Home/Desktop/esp-idf/components/lwip/port/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Home/Desktop/esp-idf/components/vfs/include -IC:/Users/Home/Desktop/esp-idf/components/esp_wifi/include -IC:/Users/Home/Desktop/esp-idf/components/esp_wifi/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/esp_eth/include -IC:/Users/Home/Desktop/esp-idf/components/efuse/include -IC:/Users/Home/Desktop/esp-idf/components/efuse/esp32/include -IC:/Users/Home/Desktop/esp-idf/components/app_trace/include -IC:/Users/Home/Desktop/esp-idf/components/mbedtls/port/include -IC:/Users/Home/Desktop/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/Home/Desktop/esp-idf/components/wpa_supplicant/include -IC:/Users/Home/Desktop/esp-idf/components/wpa_supplicant/port/include -IC:/Users/Home/Desktop/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/Users/Home/Desktop/esp-idf/components/bootloader_support/include -IC:/Users/Home/Desktop/esp-idf/components/app_update/include -IC:/Users/Home/Desktop/esp-idf/components/spi_flash/include -IC:/Users/Home/Desktop/esp-idf/components/nvs_flash/include -IC:/Users/Home/Desktop/esp-idf/components/pthread/include -IC:/Users/Home/Desktop/esp-idf/components/espcoredump/include -IC:/Users/Home/Desktop/esp-idf/components/asio/asio/asio/include -IC:/Users/Home/Desktop/esp-idf/components/asio/port/include -IC:/Users/Home/Desktop/esp-idf/components/coap/port/include -IC:/Users/Home/Desktop/esp-idf/components/coap/port/include/coap -IC:/Users/Home/Desktop/esp-idf/components/coap/libcoap/include -IC:/Users/Home/Desktop/esp-idf/components/coap/libcoap/include/coap2 -IC:/Users/Home/Desktop/esp-idf/components/console -IC:/Users/Home/Desktop/esp-idf/components/nghttp/port/include -IC:/Users/Home/Desktop/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/Home/Desktop/esp-idf/components/esp-tls -IC:/Users/Home/Desktop/esp-idf/components/esp_adc_cal/include -IC:/Users/Home/Desktop/esp-idf/components/esp_gdbstub/include -IC:/Users/Home/Desktop/esp-idf/components/tcp_transport/include -IC:/Users/Home/Desktop/esp-idf/components/esp_http_client/include -IC:/Users/Home/Desktop/esp-idf/components/esp_http_server/include -IC:/Users/Home/Desktop/esp-idf/components/esp_https_ota/include -IC:/Users/Home/Desktop/esp-idf/components/protobuf-c/protobuf-c -IC:/Users/Home/Desktop/esp-idf/components/protocomm/include/common -IC:/Users/Home/Desktop/esp-idf/components/protocomm/include/security -IC:/Users/Home/Desktop/esp-idf/components/protocomm/include/transports -IC:/Users/Home/Desktop/esp-idf/components/mdns/include -IC:/Users/Home/Desktop/esp-idf/components/esp_local_ctrl/include -IC:/Users/Home/Desktop/esp-idf/components/esp_websocket_client/include -IC:/Users/Home/Desktop/esp-idf/components/expat/expat/expat/lib -IC:/Users/Home/Desktop/esp-idf/components/expat/port/include -IC:/Users/Home/Desktop/esp-idf/components/wear_levelling/include -IC:/Users/Home/Desktop/esp-idf/components/sdmmc/include -IC:/Users/Home/Desktop/esp-idf/components/fatfs/diskio -IC:/Users/Home/Desktop/esp-idf/components/fatfs/vfs -IC:/Users/Home/Desktop/esp-idf/components/fatfs/src -IC:/Users/Home/Desktop/esp-idf/components/freemodbus/common/include -IC:/Users/Home/Desktop/esp-idf/components/idf_test/include -IC:/Users/Home/Desktop/esp-idf/components/jsmn/include -IC:/Users/Home/Desktop/esp-idf/components [... cut ...]
../main/chat_server.cpp:192:13: warning: 'void got_ip_event_handler(void*, esp_event_base_t, int32_t, void*)' defined but not used [-Wunused-function]
 static void got_ip_event_handler(void* arg, esp_event_base_t event_base,
             ^~~~~~~~~~~~~~~~~~~~
../main/chat_server.cpp:152:13: warning: 'void wifi_event_handler(void*, esp_event_base_t, int32_t, void*)' defined but not used [-Wunused-function]
 static void wifi_event_handler(void* arg, esp_event_base_t event_base,
             ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
Build complete (20 errors, 6 warnings): C:\Users\Home\eclipse-workspace\asio1\build

krzychb, 2020-03-04:
>if I remove component directories from a CMakelists file, then a lot of errors appear during compilation.

> Example my CMakelists:
> cmake_minimum_required(VERSION 3.5)
> set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
> project(asio_chat_server)

@komandor2 unmodified example `examples/protocols/asio/chat_server/` compiles correctly for me on master branch, both in Eclipse and from command line. 

1. Does unmodified example compile correctly for you?
2. What exact "component directories" do you remove from this example?
3. How do you adjust this example to ensure that removed "component directories" are not required anymore?

renzbagaporo, 2020-03-05:
@komandor2  From the error log you posted, it seems that you followed the recommendation of picking an example as a base and copy-pasting code from other examples?

It seems to me that there might be some stuff missing or some errors with copy-pasting. Examples:

1. `../main/chat_server.cpp:103:14: error: 'TAG' was not declared in this scope`

Looks to me that this was copied from an example that uses `ESP_LOG` functions. It's missing the tag, which is usually provided by a `#define TAG` or a `static const char* TAG = `. Also, to be able to use these API, you need `esp_log.h`.

2.` ../main/chat_server.cpp:152:13: warning: 'void wifi_event_handler(void*, esp_event_base_t, int32_t, void*)' defined but not used [-Wunused-function]`

This is just a warning, but just note that copying this does not mean the `wifi_event_handler` works since nothing is actually using it.

~~~~~~

I recommend copying stuff incrementally and verifying stuff works before moving on to the next functionality needed.

komandor2, 2020-03-05:
Thanks for the answer, the esp_log.h is included, I think there is an error due to the repeated announcement. By the second warning, I understand.

renzbagaporo, 2020-03-11:
Can this be closed @komandor2?

closed 2020-03-11 by komandor2