espressif/esp-idf #4872: esp_http_client_close always returns -1 on an https request

sebirdman, 2020-03-02:
## Problem Description

in the current 3.3 stable branch and master branches esp_http_client_close always returns ESP_FAIL. The errno is always 9. 

This appears to be a result of transport_ssl.c's implementation of ssl_close where the return value is always -1

https://github.com/espressif/esp-idf/blob/aa3b1da384e78c769d7f8a3c2bf3f24a56a8df29/components/tcp_transport/transport_ssl.c#L168
```
static int ssl_close(esp_transport_handle_t t)
{
    int ret = -1;
    transport_ssl_t *ssl = esp_transport_get_context_data(t);
    if (ssl->ssl_initialized) {
        esp_tls_conn_delete(ssl->tls);
        ssl->ssl_initialized = false;
    }
    return ret;
}
``` 

because the current implementation of esp_http_client_close always returns the result of esp_transport_close.

This conflicts with the names of the returned values according to the docs: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_client.html#_CPPv421esp_http_client_close24esp_http_client_handle_t

AxelLin, 2020-03-02:
The same issue on git master tree:
https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_ssl.c#L168
https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_tcp.c#L162

david-cermak, 2020-03-05:
Thanks @sebirdman for posting this issue. Seems like an oversight which needs fixing.

closed 2020-05-07 by espressif-bot