espressif/esp-idf #4873: 运行数小时后，idf崩溃了

fzboffice, 2020-03-03:
运行数小时后，idf崩溃了，错误报告如下
idf 版本是4.0 f9cb434e

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084891  PS      : 0x00060834  A0      : 0x80081faf  A1      : 0x3ffaf9c0  
0x40084891: esp_timer_impl_set_alarm at /home/fzb/esp/esp-idf/components/esp32/esp_timer_esp32.c:239

A2      : 0x13000236  A3      : 0x00000005  A4      : 0x3ffd0470  A5      : 0x3ffd046c  
A6      : 0x000000a0  A7      : 0x00000001  A8      : 0xefffffff  A9      : 0x3ff47030  
A10     : 0xeffffffe  A11     : 0xf25b94e2  A12     : 0xf00137e0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060023  SAR     : 0x0000000e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 977660c726facaf400ca3c411ffbaf9cd049c2d4136e2e3d1cb5f5a041f2dbaa

Backtrace: 0x4008488e:0x3ffaf9c0 0x40081fac:0x3ffaf9f0 0x400d4acc:0x3ffafa10
0x4008488e: esp_timer_impl_set_alarm at /home/fzb/esp/esp-idf/components/esp32/esp_timer_esp32.c:238

0x40081fac: timer_insert at /home/fzb/esp/esp-idf/components/esp_common/src/esp_timer.c:204

0x400d4acc: timer_process_alarm at /home/fzb/esp/esp-idf/components/esp_common/src/esp_timer.c:284
 (inlined by) timer_task at /home/fzb/esp/esp-idf/components/esp_common/src/esp_timer.c:318

Core 1 register dump:
PC      : 0x40177aca  PS      : 0x00060234  A0      : 0x800d4c58  A1      : 0x3ffb2f10  
0x40177aca: lwip_standard_chksum at /home/fzb/esp/esp-idf/components/lwip/lwip/src/core/inet_chksum.c:151

A2      : 0x00000000  A3      : 0x40001d48  A4      : 0x00060e20  A5      : 0x00000001  
A6      : 0x0000abab  A7      : 0x00000000  A8      : 0x800d4776  A9      : 0x3ffb2ee0  
A10     : 0x00000000  A11     : 0x3ffbede8  A12     : 0x3ffbede8  A13     : 0x00000000  
A14     : 0x00060e20  A15     : 0x3ffeaa00  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 977660c726facaf400ca3c411ffbaf9cd049c2d4136e2e3d1cb5f5a041f2dbaa

Backtrace: 0x40177ac7:0x3ffb2f10 0x400d4c55:0x3ffb2f30 0x400903ac:0x3ffb2f50
0x40177ac7: lwip_standard_chksum at /home/fzb/esp/esp-idf/components/lwip/lwip/src/core/inet_chksum.c:150

0x400d4c55: esp_vApplicationIdleHook at /home/fzb/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400903ac: prvIdleTask at /home/fzb/esp/esp-idf/components/freertos/tasks.c:3382

ESP-Coco, 2020-03-04:
中断看门狗复位，从你的 backtrace 推测你在 timer函数里面有做什么操作导致的，一般来说是存在延时，当然也不排除其他内存踩踏问题

fzboffice, 2020-03-06:
感谢

closed 2020-03-06 by fzboffice