espressif/esp-idf #4874: SHA-256 for current firmware issue

sahilsoni1, 2020-03-03:
0x40081480: call_start_cpu1 at /home/sahilsoni/esp/esp-idf/components/esp32/cpu_start.c:274

I (0) cpu_start: App cpu up.
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (530) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (536) heap_init: At 3FFB98D8 len 00026728 (153 KiB): DRAM
I (543) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (555) heap_init: At 40095840 len 0000A7C0 (41 KiB): IRAM
I (562) cpu_start: Pro cpu start user code
I (580) spi_flash: detected chip: gd
I (580) spi_flash: flash io: dio
I (581) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (591) native_ota_example: SHA-256 for the partition table: : d1c0e9d02fa9d26cd2e1984e7b5dd20157204f501ddc83ce82229e5f3175ee8b
I (621) native_ota_example: SHA-256 for bootloader: : 6e0d03b4e66823c4dcd681b7790996e4915f0e3ca9cde7616a340425bbd33041

/////////////////////////////////////////////////////////////////////////////////////////*
file: native_ota.bin
file **size:8,48,656bytes
SHA-256native_ota.bin:feba2f6b5301b4c46c99237b2d6ce4684dd17ca4c5a014833db0b616736cdfc7
*///////////////////////////////////////////////////////////////////////////////////////

label of running app:factory
address of running app:65536
size of running app:1048576

I (911) native_ota_example: SHA-256 for current firmware: : afa366e53e86c32eeac1402c6c4c7a59c083fddeaa488f1edbadfd14ab8367e3
/////////////////////////////////////////////////////////////////////////////////////////////////*
if i am comparing theSHA-256  current firmware with native_ota.bin file it is not matching.And also size difference between current firmware(1048576) and native_ota.bin(848656)
may be reason is code take whole size of allocated space factory app but burn file is smaller then that.
*////////////////////////////////////////////////////////////////////

projectgus, 2020-04-01:
Hi @sahilsoni1 ,

Sorry for the delay in getting back to you.

The printout of "SHA-256 for current firmware" is based on the result of the [esp_ota_get_running_partition()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#_CPPv424esp_partition_get_sha256PK15esp_partition_tP7uint8_t) function is the value of a SHA-256 hash appended to the firmware file itself, and used for verification.

To print this value locally, run something like:

```
esptool.py --chip esp32 image_info build/firmware.bin
```

... the value printed here should match, but the SHA-256 of the entire .bin file will not be the same (because there is a SHA-256 appended to the .bin file itself.)

More details can be found at the documentation link, above.

> And also size difference between current firmware(1048576) and native_ota.bin(848656)
> may be reason is code take whole size of allocated space factory app but burn file is smaller then that.

The `native_ota` example doesn't print this information, so the answer depends on what code was added to print it. If you're printing `partition.size` then this is the size of the app partition, not the size of the app binary.

To get a more detailed answer, please provide the code which prints this line and also the exact ESP-IDF version that you're using (this is printed in the boot log, usually.)

Alvin1Zhang, 2020-05-13:
@sahilsoni1 Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

projectgus, 2020-06-03:
Closing due to a lack of updates.

closed 2020-06-03 by projectgus