espressif/esp-idf #4875: dns provision in udp_client()

sumit809, 2020-03-03:
So, there is only one option of providing IPv4 address in Example Configuration in menuconfig. As the IP may change, can there be an option to enter DNS directly?

david-cermak, 2020-03-05:
Hi @sumit809 

Do you mean to use a host name instead of IP in examples? Yes it is possible, of course, but the socket examples are meant so demonstrate very simple socket manipulations and adding a dns resolver might add unnecessary complexity.  Moreover one can check other examples that connect to hostname rather than ip address. Linking the `esp-tls` component which is used by many other protocol examples as a reference: https://github.com/espressif/esp-idf/blob/master/components/esp-tls/esp_tls.c#L128

sumit809, 2020-03-11:
Okay @david-cermak !
I will look into it!
Thanks...! :)

closed 2020-03-11 by sumit809