espressif/esp-idf #4876: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1) When stress test SPI external flash

FrGrQuim, 2020-03-03:
## Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):
    // v4.0-rc
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2.0
- Operating System:     Linux
- Power Supply:        external 5V

## Problem Description

We develop a station (base on a esp32) for a health  bracelet. The station get the bracelet data via BLE, manage the data and send it via mqtt on a server. When the wifi/mqtt isn't connected, the data are saved in a external flash memory connected via SPI. 

The station is also able to play music. The music are also saved in the SPI external flash. When music is played, a task will read a fix number of sample and sand it to the audio codec via i2s. when the i2s buffer is send, the task will read the next sample, etc. 

A semaphore was added to avoid risque of multi-call to spi_transfert function (take just before, release just after)

We do a stress test of the SPI extrernal flash by loop playing music with a bracelet connected bu no wifi (data are written in SPI external flash).  

### Expected Behavior
The esp32 is able to write (ble data) and read (play music) in the same time without crashing

### Actual Behavior
After several hours, the ESP32 stop working (Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1) )

## Debug Logs

periodic_task  	232705839		5%
music_play_task	899168351		22%
Tmr Svc        	5346569		<1%
IDLE1          	3546688256		88%
IDLE0          	2870422359		71%
i2s_read_task  	1064949000		26%
band_tmp_data_t	308455069		7%
tiT            	43113710		1%
main           	585483		<1%
ipc1           	214275		<1%
ipc0           	92902		<1%
uart_event_task	50		<1%
sys_evt        	9512388		<1%
mqtt_log_task  	742794		<1%
httpd          	1617		<1%
wifi           	280604090		7%
hciT           	1565494113		39%
btuT           	1346213055		33%
BTC_TASK       	538554129		13%
esp_timer      	84063093		2%
btController   	4093607157		102%
dl_task        	103		<1%
update_manager 	30		<1%
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40090c56  PS      : 0x00060034  A0      : 0x80099d79  A1      : 0x3f8379f0  
A2      : 0x3f837078  A3      : 0x3f40bf38  A4      : 0x00000014  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x3f40bf38  A9      : 0x3f837078  
A10     : 0x00000001  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x3f40bb5e  
A14     : 0x3ff47024  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400932a9  LEND    : 0x400932b9  LCOUNT  : 0xfffffff8  

## Other items if possible

[spi_stress_test2.txt](https://github.com/espressif/esp-idf/files/4280879/spi_stress_test2.txt)

Dazza0, 2020-03-09:
@FrGrQuim 

Since you are stress testing, it could very well be that the run time of the interrupt handlers are too long. `btController` having a CPU usage of `102%` suggests to me that the ESP32 CPU is being heavily loaded, thus it's likely a lot of interrupts are happening quite frequently.

- I suggest increasing the `CONFIG_ESP_INT_WDT_TIMEOUT_MS` to something like `500ms` and see if the issue persists.

- You could also try distribute your interrupts evenly across core 0 and core 1. Most of the WiFi and BT interrupts run on core 0. So if you have any other application related interrupts, try to put them on core 1, so that core 0 isn't spending too much time handling interrupts.

- You could also run addr2line on `0x40090c56` to see what function/ISR was being run when the INT WDT triggered.

Alvin1Zhang, 2020-08-18:
@FrGrQuim Thanks for reporting, would you please help check if any updates for the issue? Thanks.

bioshock2k, 2020-10-09:
I have a similar problem! I am wondering is how you did the logs with the percentages?

ginkgm, 2021-08-17:
Hi @FrGrQuim 
One thing to clarify, what do you mean by "external flash"? Which peripheral (or pins) is it connected to?

When we refer to "external flash", we usually means the flash which doesn't used for holding the code, and sometimes connected to other SPI buses like HSPI (SPI2) /VSPI (SPI3). 
If I remember correctly, the external flash driver on your IDF version doesn't involve the ISR, so it shouldn't cause the INT WDT timeout. 

Could you provide more information to help me clarify the issue?

AxelLin, 2021-08-18:
> @FrGrQuim
> 
> Since you are stress testing, it could very well be that the run time of the interrupt handlers are too long. `btController` having a CPU usage of `102%` 

I think that CPU usage is just wrong.

IDLE1 3546688256 88%
IDLE0 2870422359 71%
btController 4093607157 102%

ginkgm, 2022-02-19:
Issue closed due to no reply. Feel free to re-open it if you have more questions.

closed 2022-02-19 by ginkgm